koide3/hdl_graph_slam #3: how to calibrate the coordinate between GPS and slam

weisongwen, 2018-03-19:
Dear @koide3 

you mentioned that you employ the GPS into the slam, but how do you calibrate the coordinate between GPS and slam?

Best 
Welson

koide3, 2018-03-21:
Hi,
In this package, GPS data (latitude and longitude) are converted into the UTM coordinate (X and Y), 
and then, they are associated with pose nodes with the closest timestamps. They are optimized so that the relative position error in the UTM coordinate is minimized.

Since our GPS-based optimization is based on relative position information, we do not explicitly calibrate the map coordinate and the (global) GPS coordinate.

weisongwen, 2018-03-21:
hi @koide3 
Thanks for your kind reply,

koide3, 2018-03-22:
Hi,
Actually, our implementation of the GPS-based optimization is a very naive way. It assumes that the GPS and the LIDAR shared the same coordinate, and doesn't consider the bias between them. But, I suppose, in most of the cases, the GPS error much more affects the mapping accuracy, and the lack of consideration of the bias doesn't really matter.
On the other hand, adding the bias term to the map optimization is not so hard. If you face any problem due to the bias, we can modify the package to deal with the bias.

weisongwen, 2018-03-22:
Hi @koide3 
Thanks for your kind reply, actually, the bias between the GPS coordinate system and mapping coordinate system is unknown, but we can get this bias by the magnetometer sensor? or you have some algorithm to estimate the bias? 

Thanks 
Best 
Welson

weisongwen, 2018-03-27:
Hi @koide3  
do you have any paper published that we can referred to about this package?

Best 
Welson

closed 2018-04-17 by koide3

weisongwen, 2018-06-14:
hi @koide3 
I refer to the code adding the gps into the graph optimization. 
`Eigen::Vector3d xyz(utm.easting, utm.northing, utm.altitude); // save the 

      if(!zero_utm) { // initial xyz
        zero_utm = xyz;
      }
      xyz -= (*zero_utm);`
in this  code , the xyz represent the motion difference in the north direction, east direction and altitude direction respectively?
best 
welson

koide3, 2018-06-15:
@weisongwen ,
Yes, that xyz represents the motion difference from the first node in the [UTM coordinate](http://wiki.ros.org/geodesy). Currently, altitude is ignored, but if you want, you can take it into account by changing [here](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L278).

weisongwen, 2018-06-17:
Dear @koide3 ,
thanks for your kind information, 
best 
welson

cslxiao, 2018-12-14:
@koide3 
The code crashes when loop is detected:
`
 --- loop detection ---
num_candidates: 1
matchingdouble free or corruption (out)
[velodyne_nodelet_manager-5] process has died [pid 2418, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/leon/.ros/log/62613666-ff9c-11e8-927c-c0b6f992692c/velodyne_nodelet_manager-5.log].`

liuqi8827, 2019-03-28:
@koide3 
Hi, thanks for your great work.
Do you have any suggestion about how to **calibrate the map coordinate and the (global) GPS coordinate**.

Best