koide3/hdl_graph_slam #4: hdl_graph_slam hdl_graph_slam_xxx.launch FAILED

sunliamm, 2018-04-11:
Hi guys,

I got this error when I tried to launch the "hdl_graph_slam_xxx.launch" files (on both 400 and 501 bags, the g2o version is the one mentioned on the README file)

*******************************************************
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': free(): invalid next size (fast): 0x0000000001ea6c50 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f8f66fa97e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f8f66fb237a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f8f66fb653c]
/opt/ros/kinetic/lib/libroscpp.so(_ZNK3ros10NodeHandle11resolveNameERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEbNS0_11no_validateE+0xa3)[0x7f8f68413a73]
/opt/ros/kinetic/lib/libroscpp.so(_ZNK3ros10NodeHandle11resolveNameERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x6b)[0x7f8f68413d0b]
/opt/ros/kinetic/lib/libroscpp.so(_ZNK3ros10NodeHandle8hasParamERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x2c)[0x7f8f684146dc]
/home/z/iliad_ws/devel/lib//libhdl_graph_slam_nodelet.so(_ZNK3ros10NodeHandle5paramIdEEbRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERT_RKSA_+0x1b)[0x7f8f400c295b]
/home/z/iliad_ws/devel/lib//libhdl_graph_slam_nodelet.so(_ZN14hdl_graph_slam19HdlGraphSlamNodelet6onInitEv+0x47b)[0x7f8f400eb8eb]
/opt/ros/kinetic/lib/libnodeletlib.so(_ZN7nodelet7Nodelet4initERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIS7_S6_EEERKSt6vectorIS6_SaIS6_EEPN3ros22CallbackQueueInterfaceESP_+0x39e)[0x7f8f68b1623e]
/opt/ros/kinetic/lib/libnodeletlib.so(_ZN7nodelet6Loader4loadERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_RKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIS7_S6_EEERKSt6vectorIS6_SaIS6_EE+0x3a2)[0x7f8f68b183c2]
/opt/ros/kinetic/lib/libnodeletlib.so(_ZN7nodelet9LoaderROS11serviceLoadERNS_19NodeletLoadRequest_ISaIvEEERNS_20NodeletLoadResponse_IS2_EE+0x110)[0x7f8f68b26750]
/opt/ros/kinetic/lib/libnodeletlib.so(_ZN3ros22ServiceCallbackHelperTINS_11ServiceSpecIN7nodelet19NodeletLoadRequest_ISaIvEEENS2_20NodeletLoadResponse_IS4_EEEEE4callERNS_31ServiceCallbackHelperCallParamsE+0x707)[0x7f8f68b2f1f7]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros15ServiceCallback4callEv+0x241)[0x7f8f683b0351]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue9callOneCBEPNS0_3TLSE+0x4b8)[0x7f8f684046f8]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue13callAvailableENS_12WallDurationE+0x54b)[0x7f8f684060fb]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros21SingleThreadedSpinner4spinEPNS_13CallbackQueueE+0x2a9)[0x7f8f68462ef9]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros4spinEv+0x2b)[0x7f8f68447edb]
/opt/ros/kinetic/lib/nodelet/nodelet(main+0x1e7)[0x405127]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f8f66f52830]
/opt/ros/kinetic/lib/nodelet/nodelet(_start+0x29)[0x406449]
======= Memory map: ========
00400000-0040e000 r-xp 00000000 103:02 47320383                          /opt/ros/kinetic/lib/nodelet/nodelet
0060d000-0060e000 r--p 0000d000 103:02 47320383                          /opt/ros/kinetic/lib/nodelet/nodelet
0060e000-0060f000 rw-p 0000e000 103:02 47320383                          /opt/ros/kinetic/lib/nodelet/nodelet
01b57000-0224b000 rw-p 00000000 00:00 0                                  [heap]
7f8f30000000-7f8f30021000 rw-p 00000000 00:00 0 
7f8f30021000-7f8f34000000 ---p 00000000 00:00 0 
7f8f3522a000-7f8f35233000 r-xp 00000000 103:02 11543024                  /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7f8f35233000-7f8f35432000 ---p 00009000 103:02 11543024                  /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7f8f35432000-7f8f35433000 r--p 00008000 103:02 11543024                  /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7f8f35433000-7f8f35434000 rw-p 00009000 103:02 11543024                  /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7f8f35434000-7f8f3543e000 r-xp 00000000 103:02 11540753                  /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0
7f8f3543e000-7f8f3563d000 ---p 0000a000 103:02 11540753                  /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0
7f8f3563d000-7f8f3563e000 r--p 00009000 103:02 11540753                  /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0
7f8f3563e000-7f8f3563f000 rw-p 0000a000 103:02 11540753                  /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0
7f8f3563f000-7f8f35644000 r-xp 00000000 103:02 11542241                  /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f8f35644000-7f8f35843000 ---p 00005000 103:02 11542241                  /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f8f35843000-7f8f35844000 r--p 00004000 103:02 11542241                  /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f8f35844000-7f8f35845000 rw-p 00005000 103:02 11542241                  /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f8f35845000-7f8f35847000 r-xp 00000000 103:02 11542230                  /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f8f35847000-7f8f35a47000 ---p 00002000 103:02 11542230                  /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f8f35a47000-7f8f35a48000 r--p 00002000 103:02 11542230                  /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f8f35a48000-7f8f35a49000 rw-p 00003000 103:02 11542230                  /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f8f35a49000-7f8f35a82000 r-xp 00000000 103:02 11548756                  /usr/lib/x86_64-linux-gnu/libhwloc.so.5.6.8
7f8f35a82000-7f8f35c81000 ---p 00039000 103:02 11548756                  /usr/lib/x86_64-linux-gnu/libhwloc.so.5.6.8
7f8f35c81000-7f8f35c82000 r--p 00038000 103:02 11548756                  /usr/lib/x86_64-linux-gnu/libhwloc.so.5.6.8
7f8f35c82000-7f8f35c83000 rw-p 00039000 103:02 11548756                  /usr/lib/x86_64-linux-gnu/libhwloc.so.5.6.8
7f8f35c83000-7f8f35ca4000 r-xp 00000000 103:02 11543648                  /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7f8f35ca4000-7f8f35ea3000 ---p 00021000 103:02 11543648                  /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7f8f35ea3000-7f8f35ea4000 r--p 00020000 103:02 11543648                  /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7f8f35ea4000-7f8f35ea5000 rw-p 00021000 103:02 11543648                  /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7f8f35ea5000-7f8f35eb0000 r-xp 00000000 103:02 11542978                  /usr/lib/x86_64-linux-gnu/libjbig.so.0
7f8f35eb0000-7f8f360af000 ---p 0000b000 103:02 11542978                  /usr/lib/x86_64-linux-gnu/libjbig.so.0
7f8f360af000-7f8f360b0000 r--p 0000a000 103:02 11542978                  /usr/lib/x86_64-linux-gnu/libjbig.so.0
7f8f360b0000-7f8f360b3000 rw-p 0000b000 103:02 11542978                  /usr/lib/x86_64-linux-gnu/libjbig.so.0
7f8f360b3000-7f8f360d4000 r-xp 00000000 103:02 48894453                  /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7f8f360d4000-7f8f362d3000 ---p 00021000 103:02 48894453                  /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7f8f362d3000-7f8f362d4000 r--p 00020000 103:02 48894453                  /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7f8f362d4000-7f8f362d5000 rw-p 00021000 103:02 48894453                  /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7f8f362d5000-7f8f36365000 r-xp 00000000 103:02 11548784                  /usr/lib/openmpi/lib/libopen-pal.so.13.0.2
7f8f36365000-7f8f36565000 ---p 00090000 103:02 11548784                  /usr/lib/openmpi/lib/libopen-pal.so.13.0.2
7f8f36565000-7f8f36569000 r--p 00090000 103:02 11548784                  /usr/lib/openmpi/lib/libopen-pal.so.13.0.2
[... 519 lines cut ...]
7f8f68f43000-7f8f68f44000 rw-p 00000000 00:00 0 
7f8f68f44000-7f8f68f62000 r-xp 00000000 103:02 48890201                  /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8f68f62000-7f8f68f63000 r--p 0001d000 103:02 48890201                  /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8f68f63000-7f8f68f64000 rw-p 0001e000 103:02 48890201                  /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8f68f64000-7f8f68f77000 rw-p 00000000 00:00 0 
7f8f68f77000-7f8f68f78000 r--p 00025000 103:02 48889873                  /lib/x86_64-linux-gnu/ld-2.23.so
7f8f68f78000-7f8f68f79000 rw-p 00026000 103:02 48889873                  /lib/x86_64-linux-gnu/ld-2.23.so
7f8f68f79000-7f8f68f7a000 rw-p 00000000 00:00 0 
7fff42482000-7fff424a4000 rw-p 00000000 00:00 0                          [stack]
7fff42518000-7fff4251b000 r--p 00000000 00:00 0                          [vvar]
7fff4251b000-7fff4251d000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[... cut ...]
[31m[hdl_graph_slam_nodelet-5] process has died [pid 504, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/z/.ros/log/40bdce56-3da1-11e8-ab99-28c63f080efd/hdl_graph_slam_nodelet-5.log].
log file: /home/z/.ros/log/40bdce56-3da1-11e8-ab99-28c63f080efd/hdl_graph_slam_nodelet-5*.log[0m
************************************************

Any ideas will be appreciated.
Many thanks in advance.
Kevin

koide3, 2018-04-17:
Hi, 

I found that some SSE options cause the error on some PCs. Can you remove the options from CMakeList.txt, and re-build the package after removing "devel" and "build" directories?

Change:
```txt:CMakeLists.txt
add_definitions(-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2)
set(CMAKE_CXX_FLAGS "-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2")
```
to:
```txt:CMakeLists.txt
add_definitions(-std=c++11)
set(CMAKE_CXX_FLAGS "-std=c++11)
```
and:
```bash
roscd
cd ..
# Do it carefully
rm -rf build devel
catkin_make
```

I suspect I made some Eigen-related bad codes. I'll try to understand what actually causes this problem.

sunliamm, 2018-04-17:
Thanks a lot for your reply!

I did that, while the issue is not solved. I got this (P.S. I also modifies the same in the CMAKELIST.txt in ndt_omp package):

  [vsyscall]
[FATAL] [1523980167.100801181]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1523980167.100886308]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 26459, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/velodyne_nodelet_manager-2.log].
log file: /home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/velodyne_nodelet_manager-2*.log
[scan_matching_odometry_nodelet-4] process has died [pid 26461, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/scan_matching_odometry_nodelet-4.log].
log file: /home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/scan_matching_odometry_nodelet-4*.log
[hdl_graph_slam_nodelet-6] process has died [pid 26484, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/hdl_graph_slam_nodelet-6.log].
log file: /home/z/.ros/log/e816d7a6-4256-11e8-8e6a-28c63f080efd/hdl_graph_slam_nodelet-6*.log

Any ideas?

sunliamm, 2018-04-23:
> I have the same problem. Should g2o install in catkin_ws ?

i dont think so. but seems this error is from g2o...

which ros version are you using? i am using kinetic on 16.04 not sure，but
this rep seems only tested with indigo on 14.04

koide3, 2018-04-23:
You don't need to put g2o in catkin_ws. I'm testing this package on kinetic, but I didn't see the error. Can you try older g2o than the one I mentioned in README?

sunliamm, 2018-04-23:
sure. will let you know :)

thomas3016, 2018-04-30:
I met the same question, and I'm using indigo on 14.04. Have you found a soultion?

koide3, 2018-05-13:
Hi guys,

Sorry for that I didn't reply for a long time. I was a bit busy this two weeks...

I found that, in some environments, **g2o::VertexPlane** causes memory corruption. I removed a variable (named color) in [g2o/vertex_plane.h](https://github.com/RainerKuemmerle/g2o/blob/a48ff8c42136f18fbe215b02bfeca48fa0c67507/g2o/types/slam3d_addons/vertex_plane.h#L71) and [cpp](https://github.com/RainerKuemmerle/g2o/blob/a48ff8c42136f18fbe215b02bfeca48fa0c67507/g2o/types/slam3d_addons/vertex_plane.cpp#L35). After re-building g2o and hdl_graph_slam, it seemed the error was solved. 

You can remove it by yourself from g2o, or clone my [g2o repository](https://github.com/koide3/g2o) and checkout *hdl_graph_slam* branch. It is identical to the commit *a48ff8c42136f18fbe215b02bfeca48fa0c67507* of g2o except that the color field was removed.

I also made a docker environment for this package. If you still see this problem, try it. See README for details.

Xiangzhaohong, 2018-05-16:
@koide3 Hi, wo already solve this problem after I change the g2o which checkout hdl_graph_slam branch.
But I meet another problem when I play rosbag:
optimizing... [velodyne_nodelet_manager-1] process has died [pid 24941, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/velodyne_nodelet_manager-1.log].
log file: /home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/velodyne_nodelet_manager-1*.log
[floor_detection_nodelet-4] process has finished cleanly
log file: /home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/floor_detection_nodelet-4*.log
[hdl_graph_slam_nodelet-5] process has finished cleanly
log file: /home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/hdl_graph_slam_nodelet-5*.log
[prefiltering_nodelet-2] process has finished cleanly
log file: /home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/prefiltering_nodelet-2*.log
[scan_matching_odometry_nodelet-3] process has finished cleanly
log file: /home/shuai/.ros/log/36ed214c-5908-11e8-b2e2-d46a6a561f99/scan_matching_odometry_nodelet-3*.log

What should I do to deal with it?
Thank u very much!

koide3, 2018-05-16:
OK, recently, I created a docker environment for this package. Can you try it? You can find the instruction in README. I think we don't see any dependency problem on docker...

bugraturan, 2018-05-23:
Dear @koide3 I have the same error as @Xiangzhaohong even in the Docker environment. By the way in the README there is a second roslaunch in the Docker environment instructions :-)

Thanks and Cheers

koide3, 2018-05-25:
@bugraturan 
Hmmm, It's hard to figure out what causes the error since I cannot reproduce it... I guess it occurs only on some hardware. Can you tell me your software and hardware information? (Ubuntu and ROS versions, CPU, memory, etc.)

I'll fix the instruction in the README. Thank you.

Xiangzhaohong, 2018-05-29:
@chrisy-d1989 I think it's still caused by g2o. Because it can not optimize.

neale, 2018-06-11:
+1 I have the same issue in the docker container. 
When I run on the outside data, or on my own. The optimization component crashes with 

```
optimizing... [velodyne_nodelet_manager-1] process has died [pid 1581, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/root/.ros/log/47ce4a5e-6cef-11e8-9f72-4ccc6a926a8c/velodyne_nodelet_manager-1.log].
log file: /root/.ros/log/47ce4a5e-6cef-11e8-9f72-4ccc6a926a8c/velodyne_nodelet_manager-1*.log
```

Inside data however, works just fine. 

I'm on an Intel Skylake, ubuntu 16, ros-kinetic.

StanleyHusai, 2018-06-12:
I changed my Ubuntu 16.04 ROS(kinetic) to 14.04 ROS(jade), and all the problems are gone.

koide3, 2018-06-12:
The failure happens on only kinetic? 

@Guys failed to run hdl_graph_slam, 
Can you change the ROS version in Dockerfile to indigo, and check if it works on indigo?

saimanoj18, 2018-06-13:
I have tried hdl_graph_slam on Ubuntu 16.04 and Kinetic and ended with the same issue as @neale 

`optimizing... [velodyne_nodelet_manager-1] process has died [pid 1581, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/root/.ros/log/47ce4a5e-6cef-11e8-9f72-4ccc6a926a8c/velodyne_nodelet_manager-1.log].
log file: /root/.ros/log/47ce4a5e-6cef-11e8-9f72-4ccc6a926a8c/velodyne_nodelet_manager-1*.log`

I am trying to use his gist of modified Dockerfile. 

@koide3 Can you fix it to work with Ubuntu 16.04 and Kinetic out of the box ? What can the issue ??

koide3, 2018-06-13:
@saimanoj18 
I'm not sure, but I guess some SSE-related optimizations in Eigen causes this problem. This seems do not happen on every PC since in my PC (and my colleagues' PCs), we are able to run it on ubuntu16.04 & ROS kinetic. It makes it hard to figure out the cause... I'll check the differences between the libraries in indigo and kinetic...

saimanoj18, 2018-06-13:
@koide3 I have tried on another laptop with ubuntu 16.04 and ended up with the same issues. It is possible that these are arising due to some library optimizations. Hope we can fix this!

chrisy-d1989, 2018-06-13:
@koide3 @neale I've tried it with your Dockerfile and updating cmake, but I am still getting the error in the Docker environment.

koide3, 2018-06-13:
@saimanoj18  @chrisy-d1989 
Let me make it sure that you tried indigo on docker, right?

saimanoj18, 2018-06-14:
When I try Indigo on Docker, then I could run the hdl_graph_slam_400.launch without any problem. However, the rosservice is returning False as shown below. 
`root@desay:/home# rosservice call /hdl_graph_slam/save_map 0.05 "/home/map.pcd"`
`success: False`

When I try Kinetic on the Docker, then I get the same problem of `velodyne_nodelet_manager process has died`

saimanoj18, 2018-06-14:
Using my own bagfile, I was able to save the map. So, I am planning to focus on getting hdl_graph_slam to work on Kinectic on 16.04

saimanoj18, 2018-06-19:
@koide3 @neale The error is arising from Line 159 of graph_slam.cpp file.
 `  int iterations = graph->optimize(1024);`

@koide3 Why do you think it is failing ?

koide3, 2018-06-19:
@saimanoj18 ,
I saw this error on docker with the original g2o commit (the one I forked). I tried to get rid of this error by changing a bunch of build options of g2o and eigen. But, I couldn't solve the problem, and I'm still struggling with it... I'll check the dependencies of g2o to figure out what causes this error.

saimanoj18, 2018-06-19:
@koide3 Thank you so much for your update. Let me know if you can find a solution :+1:

koide3, 2018-07-06:
I could launch the program when I built g2o with EIGEN_DONT_ALIGN_STATICALLY=1. But, it still causes segfault when optimization is performed...

koide3, 2018-07-06:
@autoperson (#17) reported that he could get rid of the error by changing the number of iterations of optimization. Can anybody try this?

koide3, 2018-07-06:
I made a new branch "devel" in which you can change the optimizer and optimization params. Can anybody try it with several optimizers and optimization params?

zyyfly2010, 2018-08-06:
@koide3 Hi,we had tested with serverl optimization params,it cound not help .

koide3, 2018-08-07:
Hi @zyyfly2010 ,
Thank you so much for the information! I'll try it on my computers...

koide3, 2018-08-07:
Hi @zyyfly2010 ,
It worked on my PC! Thanks again! Can any other guy try this way? After confirming that it works on other environments, I would like to update README.

winf, 2018-08-09:
@koide3 still doesn't work , intel i7 5550u + ubuntu16.04 + kinetic

dengwenwu1984, 2018-08-10:
Hi @koide3 @zyyfly2010 
Thanks a lot.
This way really works on my computer.

koide3, 2018-08-13:
It seems it solves the problem in most environments. Thanks again @zyyfly2010 !

@winf 
Can you try "devel" branch in which you can use several graph optimizers. Different optimizer setting might solve the error...

NZK1993, 2018-09-30:
Hi guys, How to install pcl-ros for kinetic ?  "sudo apt-get install ros-kinctic-pcl_ros" seems not right.

SimonsXuan, 2018-10-04:
> @koide3 I can help you verify and if you are in, we can try to get the hdl_graph_slam to work on Ubuntu 16.04 and Kinetic. Let me know how I can help you

Have you made it work on ubuntu 16.04?

SimonsXuan, 2018-10-04:
> Hi @koide3 @zyyfly2010
> Thanks a lot.
> This way really works on my computer.

How do you solve it? Can you show me the complete steps?

koide3, 2018-10-19:
Please try to install ros-kinetic-libg2o and build the latest version of hdl_graph_slam. It automatically detects the g2o on ROS.

LAsurvivor, 2020-07-28:
Same problem on Ubuntu 16.04:

[velodyne_nodelet_manager-2] process has died [pid 8525, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/hzh/.ros/log/182649d4-d0a0-11ea-b2f2-f875a495492a/velodyne_nodelet_manager-2.log].
log file: /home/hzh/.ros/log/182649d4-d0a0-11ea-b2f2-f875a495492a/velodyne_nodelet_manager-2*.log

I've tried every solution but didn't work, does anyone has any recommandation?

saimanoj18, 2020-07-28:
Thats probably because the velodyne driver has some issues. Try updating or running the default velodyne driver that comes default with ROS installation. I remember doing something with the velodyne driver has solved that issue.

LAsurvivor, 2020-07-28:
> Thats probably because the velodyne driver has some issues. Try updating or running the default velodyne driver that comes default with ROS installation. I remember doing something with the velodyne driver has solved that issue.

Could you show me more specific ways?

Pechpher, 2020-10-02:
@koide3 Hello, I met the same problem on Ubuntu 16.04:

ERROR: cannot launch node of type [hdl_graph_slam/map2odom_publisher.py]: hdl_graph_slam
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
[ INFO] [1601649002.363939515]: Initializing nodelet with 12 worker threads.
[ERROR] [1601649002.382224582]: Failed to load nodelet [/floor_detection_nodelet] of type [hdl_graph_slam/FloorDetectionNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class hdl_graph_slam/FloorDetectionNodelet with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet
[ERROR] [1601649002.382261051]: The error before refreshing the cache was: According to the loaded plugin descriptions the class hdl_graph_slam/FloorDetectionNodelet with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet
[FATAL] [1601649002.382398995]: Failed to load nodelet '/floor_detection_nodelet` of type `hdl_graph_slam/FloorDetectionNodelet` to manager `velodyne_nodelet_manager'
[ERROR] [1601649002.392705206]: Failed to load nodelet [/prefiltering_nodelet] of type [hdl_graph_slam/PrefilteringNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class hdl_graph_slam/PrefilteringNodelet with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFi [... cut ...]
log file: /home/pechpher/.ros/log/532f4340-04b3-11eb-b0a8-54bf64531def/floor_detection_nodelet-5*.log
[hdl_graph_slam_nodelet-6] process has died [pid 20793, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/pechpher/.ros/log/532f4340-04b3-11eb-b0a8-54bf64531def/hdl_graph_slam_nodelet-6.log].
log file: /home/pechpher/.ros/log/532f4340-04b3-11eb-b0a8-54bf64531def/hdl_graph_slam_nodelet-6*.log

wayneasimo, 2020-11-11:
> @koide3 Hi,we had tested with serverl optimization params and use your g2o branch ,it cound not help . But we found out the problem was caused by g2o or it's dependences . An good news is that the g2o Debian packages from ros can works well. We had tried on 3 computers with the same problem (start optimizing... then crashed ),it fixed the problem. Here is the setup: 1. sudo apt-get install ros-kinetic-libg2o 2. sudo cp -r /opt/ros/kinetic/lib/libg2o_* /usr/local/lib 3 . sudo cp -r /opt/ros/kinetic/include/g2o /usr/local/include 4. rebuild project( must remove the folder 'hdl_graph_slam' under catkin_ws/build beford catkin_make ). Hope it can help you!

你好，我遇到了类似问题，我用的是melodic，g2o是最新版本的，按照您的步骤并没有解决问题，请问你在melodic上有尝试吗

GoldenBoylife, 2022-10-19:
> @koide3안녕하세요, 우리는 serverl 최적화 매개변수로 테스트했고 g2o 브랜치를 사용했지만 도움이 되지 않았습니다. 그러나 우리는 문제가 g2o 또는 종속성으로 인해 발생했다는 것을 알았습니다. 좋은 소식은 ros의 g2o 데비안 패키지가 잘 작동할 수 있다는 것입니다. 우리는 같은 문제가 있는 3대의 컴퓨터에서 시도했지만(최적화 시작... 그런 다음 충돌) 문제가 해결되었습니다. 설정은 다음과 같습니다. 1. sudo apt-get install ros-kinetic-libg2o 2. sudo cp -r /opt/ros/kinetic/lib/libg2o_* /usr/local/lib 3 . sudo cp -r /opt/ros/kinetic/include/g2o /usr/local/include 4. 프로젝트 재구축( catkin_ws/build beford catkin_make 아래의 'hdl_graph_slam' 폴더를 제거해야 함). 도움이 되기를 바랍니다!

Thanks to you, I solved it well. Thank you.
To further summarize,

Remove hdl_graph_slam
-> $ catkin_make
-> Reinstall hdl_graph_slam
-> $ catkin_make
-> It's good to do roslaunch.