koide3/hdl_graph_slam #5: [ERROR] I meet some error when I launch hdl_graph_slam_xx.launch

Xiangzhaohong, 2018-04-12:
[FATAL] [1523523367.481760577]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1523523367.481761178]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 8412, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/velodyne_nodelet_manager-2.log].
log file: /home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/velodyne_nodelet_manager-2*.log
[prefiltering_nodelet-3] process has died [pid 8413, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager __name:=prefiltering_nodelet __log:=/home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/prefiltering_nodelet-3.log].
log file: /home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/prefiltering_nodelet-3*.log
[hdl_graph_slam_nodelet-6] process has died [pid 8428, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/hdl_graph_slam_nodelet-6.log].
log file: /home/shuai/.ros/log/56449ac8-3e2f-11e8-a312-d46a6a561f99/hdl_graph_slam_nodelet-6*.log

How could I solve it?
THx!

Xiangzhaohong, 2018-04-12:
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': free(): invalid next size (fast): 0x0000000000f561a0 ***

koide3, 2018-04-17:
Hi,

I guess it's the same problem as Issue #4 . Would you try to rebuild the package according to the issue?

closed 2018-05-13 by koide3