koide3/hdl_graph_slam #6: can you give me your paper?

Liying0824, 2018-04-18:
Hi，koide:

   I'm learning your open source code of hdl_graph_slam， can you give me your paper (Kenji Koide, Jun Miura, and Emanuele Menegatti, A Portable 3D LIDAR-based System for Long-term and Wide-area People Behavior Measurement, IEEE Transactions on Human-Machine Systems)?My e-mail is 85680757@qq.com
   I promise not to be used for commercial purposes, but for learning.Thanks very much.

Yours
LiYing

koide3, 2018-04-19:
Hi @Liying0824 ,
Please see the mail I sent you.

Liying0824, 2018-04-19:
Hi @koide3 ,
Thank you very much.

closed 2018-04-19 by Liying0824

frontw, 2018-06-09:
Hi!

Could you send your paper to me too? frontwise@gmail.com

Thanks!

sathya1995, 2018-06-13:
Hello @koide3 , 

I am very much interested to read your paper. Can you please send me the same to the following mail address : skasturirangan@wpi.edu  ?

Thanks a lot. 

Regards, 
Sathya Narayanan K

sysuequal, 2018-06-26:
Hello!
Could you send your paper to me too? I am learning about the 3D Lidar slam.
My email is dengyq28@mail2.sysu.edu.cn
Thanks

sunmiaozju, 2018-06-27:
Hello @koide3 ,

could you send your paper to me ,too? I am also interested in your work about 3D slam.
My email address is [sun_miao@zju.edu.cn](url)
Thanks a lot!

koide3, 2018-06-29:
Hi guys,
Please do NOT request the paper at this issue since I do not check CLOSED issues usually. In case you need the paper, send a mail to koide@aisl.cs.tut.ac.jp from your official address in your institute or organization.

lliuguangwei, 2018-08-12:
Hi @koide3  ,

I ran your code and achieved good results. Could you send your paper to me?
My email address is 285265661@qq.com

Regards,
Guangwei.Liu

yoyoyin0902, 2021-07-24:
Could you send your paper to me?
My email address is amy0937968@gmail.com
Thank you so much~