koide3/hdl_graph_slam #7: new driver Lidar

droneboxid, 2018-04-21:
Please can add the RS-16 Lidar driver to this? thanks
https://github.com/Suoivy/ros_rslidar_robosense.git

koide3, 2018-04-23:
Since I don't have the LIDAR, I can't do that. I think you can use this package with the LIDAR by just changing topic names. Would you try it and give me a merge request?

droneboxid, 2018-04-28:
hi after instal PCL and run the rviz with ROS kinetic ubuntu 16 lts. in first terminal i run 
roslaunch hdl_graph_slam hdl_graph_slam_400.launch

terminal 2 
roscd hdl_graph_slam/rviz
rviz -d hdl_graph_slam.rviz

with change cloud with rslidar_points from velodyne_points

and run with rosbag play ~

but nothing happen? the cloud topic is not detected rslidar_points

thanks

koide3, 2018-05-13:
Hi, 

Can you check if all the nodes are correctly connected using rqt_graph?

droneboxid, 2018-05-14:
hi - can you test with my sample lidar? the bag file below and the driver 
https://github.com/Suoivy/ros_rslidar_robosense.git
https://drive.google.com/file/d/1e40nZorDdVaVa0B6yTqSVHmGkHPOy5ps/view?usp=sharing

thanks

koide3, 2018-05-14:
[image]

Try the following launch file (I remapped */velodyne_points* to */rslidar_points* and changed the base frame_id of scan_matching_odometry_nodelet to "rslidar"). You can see points like in the snapshot.

```xml
<?xml version="1.0"?>
<launch>
  <!-- arguments -->
  <arg name="nodelet_manager" default="velodyne_nodelet_manager" />
  <arg name="enable_floor_detection" default="true" />
  <arg name="enable_gps" default="false" />

  <!-- in case you use velodyne_driver, comment out the following line -->
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>

  <!-- prefiltering_nodelet -->
  <node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="load hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">
  	<remap from="/velodyne_points" to="/rslidar_points"/>
    <param name="use_distance_filter" value="true" />
    <param name="distance_near_thresh" value="0.1" />
    <param name="distance_far_thresh" value="500.0" />
    <!-- NONE, VOXELGRID, or APPROX_VOXELGRID -->
    <param name="downsample_method" value="VOXELGRID" />
    <param name="downsample_resolution" value="0.1" />
    <!-- NONE, RADIUS, or STATISTICAL -->
    <param name="outlier_removal_method" value="RADIUS" />
    <param name="statistical_mean_k" value="30" />
    <param name="statistical_stddev" value="1.2" />
    <param name="radius_radius" value="0.5" />
    <param name="radius_min_neighbors" value="2" />
  </node>

  <!-- scan_matching_odometry_nodelet -->
  <node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="load hdl_graph_slam/ScanMatchingOdometryNodelet $(arg nodelet_manager)">
  	  <param name="odom_frame_id" value="odom"/>
  	  <param name="base_frame_id" value="rslidar"/>
      <param name="keyframe_delta_trans" value="1.0" />
      <param name="keyframe_delta_angle" value="1.0" />
      <param name="keyframe_delta_time" value="1.0" />
      <param name="transform_thresholding" value="true" />
      <param name="max_acceptable_trans" value="1.0" />
      <param name="max_acceptable_angle" value="1.0" />
      <param name="downsample_method" value="NONE" />
      <param name="downsample_resolution" value="0.1" />
      <!-- ICP, GICP, NDT, GICP_OMP, or NDT_OMP(recommended) -->
      <param name="registration_method" value="NDT_OMP" />
      <param name="ndt_resolution" value="1.0" />
      <param name="ndt_num_threads" value="0" />
      <param name="ndt_nn_search_method" value="DIRECT7" />
  </node>

  <!-- floor_detection_nodelet -->
  <node pkg="nodelet" type="nodelet" name="floor_detection_nodelet" args="load hdl_graph_slam/FloorDetectionNodelet $(arg nodelet_manager)" if="$(arg enable_floor_detection)">
    <param name="sensor_height" value="2.0" />
    <param name="height_clip_range" value="1.0" />
    <param name="floor_pts_thresh" value="512" />
  </node>

  <!-- hdl_graph_slam_nodelet -->
  <node pkg="nodelet" type="nodelet" name="hdl_graph_slam_nodelet" args="load hdl_graph_slam/HdlGraphSlamNodelet $(arg nodelet_manager)">
    <!-- keyframe registration params -->
    <param name="enable_gps" value="$(arg enable_gps)" />
    <param name="max_keyframes_per_update" value="10" />
    <param name="keyframe_delta_trans" value="2.0" />
    <param name="keyframe_delta_angle" value="2.0" />
    <!-- loop closure params -->
    <param name="distance_thresh" value="15.0" />
    <param name="accum_distance_thresh" value="25.0" />
    <param name="min_edge_interval" value="15.0" />
    <param name="fitness_score_thresh" value="2.5" />
    <!-- scan matching params -->
    <param name="registration_method" value="NDT_OMP" />
    <param name="ndt_resolution" value="1.0" />
    <param name="ndt_num_threads" value="0" />
    <param name="ndt_nn_search_method" value="DIRECT7" />
    <!-- information matrix params -->
    <param name="gps_edge_stddev" value="1000.0" />
    <param name="floor_edge_stddev" value="10.0" />
    <param name="use_const_inf_matrix" value="false" />
    <param name="const_stddev_x" value="0.5" />
    <param name="const_stddev_q" value="0.1" />
    <param name="var_gain_a" value="20.0" />
    <param name="min_stddev_x" value="0.1" />
    <param name="max_stddev_x" value="5.0" />
    <param name="min_stddev_q" value="0.05" />
    <param name="max_stddev_q" value="0.2" />
    <!-- update params -->
    <param name="graph_update_interval" value="3.0" />
    <param name="map_cloud_update_interval" value="10.0" />
    <param name="map_cloud_resolution" value="0.05" />
  </node>

  <node pkg="hdl_graph_slam" type="map2odom_publisher.py" name="map2odom_publisher" />
</launch>
```

droneboxid, 2018-05-15:
and can you let me know where i can find the matrix from frame? i have plan use GPS and IMU for mounting in 90 degree top down the lidar sensor. so can to mapping in the drone too. thanks

droneboxid, 2018-05-15:
i have checked the accuracy from your code is 30cm have different point with point in same location. can you make more accuracy?

koide3, 2018-05-15:
1. GPS
You can use GPS for *hdl_graph_slam*. Publish your GPS data to **/gps/geopoint** (geographic_msgs/GeoPointStamped). Try the example with GPS feature in README for the details. 

2. IMU
*hdl_graph_slam* itself does not support IMU input. If you are interested in sensor localization with IMU, see [hdl_localization](https://github.com/koide3/hdl_localization).

3. Getting positioning
You can get the transform between the map and the sensor using [tf](http://wiki.ros.org/tf/Tutorials). Like:
```bash
$ # in your case, change the frame_id "velodyne" to "rslidar"
$ rosrun tf tf_echo map velodyne
...
At time 1509348858.413
- Translation: [-4.004, 67.450, 1.599]
- Rotation: in Quaternion [-0.035, -0.032, -0.841, 0.539]
            in RPY (radian) [0.016, -0.094, -2.003]
            in RPY (degree) [0.945, -5.407, -114.753]
```
You can also subscribe **/odom** (nav_msgs/Odometry) to get the result of the scan matching.

4. Save the generated map
You can save the generated map using a service **/hdl_graph_slam/save_map**. See README for the detail.

5. Accuracy
Sorry, I don't understand what you meant. How did you measure the accuracy?

droneboxid, 2018-05-16:
Hi Mate

1. GPS
Please can you see my BAG file below? just EKF and GPS file. how can i change to /gps/geopoint ?
[https://drive.google.com/file/d/1hu9wjHYjq2dNixnMc33WRpjCt76s9xaa/view?usp=sharing](url)

2. IMU
also in the BAG file in no 1

3. Getting positioning
Please can you let me know in print screen where can i change the translation and rotation?
can i use this on real time not in replay bag file?

4. Save the generate map
I will try - thanks

5. Accuracy
below the picture of profile from the road. you can see the different point on 0 to - 0.5 m different in same section

[https://drive.google.com/file/d/1Y58Id9FN_IYl8pLJPYuJBPmIbFkyDgqc/view?usp=sharing](url)

thanks - mate

droneboxid, 2018-05-16:
[image]

koide3, 2018-05-16:
Hi, 

1. GPS and IMU
I cannot access the link. But, you can change the topic name with [remap](http://wiki.ros.org/roslaunch/XML/remap).

3. Positioning
You cannot change the output coordinate. Instead of that, you can define your new tf reference frame, and get the transform between the frame and the sensor. See [tf](http://wiki.ros.org/tf/Tutorials).

5. I think there are two reasons for the accuracy problem. 1) The sensor may not be calibrated well. 2) The error of the scan matching. I would say it is hard to get more accuracy using the 3D LIDAR and this kind of SLAM algorithms. If you need more accuracy, consider other sensors like [FARO](http://www.microgeo.it/it/laser-scanner-faro/faro-focus-s-350.aspx).

droneboxid, 2018-05-19:
Hi Thanks for advice.. please can you let me know how can i make more dense cloud? if you see on last image i show you the point cloud not much dense... thanks

koide3, 2018-05-20:
You can set a small value to [*keyframe_delta_trans* of hdl_graph_slam_nodelet](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/launch/hdl_graph_slam.launch#L62). It increases the number of keypoints (i.e., pose nodes in the graph), and thus, you can get a dense point cloud. Note that it also increases the loop closure cost.

droneboxid, 2018-05-24:
Hi it's possible to get the data from Drone? with 90 degree top down scanner position on the drone?

droneboxid, 2018-06-03:
hi - can you let me know the remap topic name for GPS?
<remap from="/gps_fix" to="/mavros/position_gps_fix"/>

like that?
thanks

droneboxid, 2018-06-03:
remap from="   ?????  " to="/mavros/positioning/gps_fix"/>

droneboxid, 2018-06-04:
please find the link below of BAG file IMU GPS and RSLIDAR. can you make code for this? many thanks

https://drive.google.com/file/d/1bQIyBldM48INX4FlYKKmQ0mx652wNZUC/view?usp=sharing

koide3, 2018-06-12:
Hi, 
You should remap */gps/geopoint* (geographic_msgs::GeoPointStamped). If it will work well, please let me know.

koide3, 2018-06-13:
The bag contains inconsistent timestamps. I saw jump back warnings on the terminal running rviz, it makes some problems on optimization and visualization.

```
[ WARN] [1528933161.230382083, 1528107517.563182079]: Detected jump back in time of 9020.62s. Clearing TF buffer.
[ WARN] [1528933161.784497767, 1528107517.837366687]: Detected jump back in time of 9020.66s. Clearing TF buffer.
[ WARN] [1528933162.698018303, 1528107518.295958061]: Detected jump back in time of 9020.76s. Clearing TF buffer.
[ WARN] [1528933164.823313715, 1528107519.359340922]: Detected jump back in time of 9020.76s. Clearing TF buffer.
```
You can confirm that the timestamp the jump back by listing to /clock topic.
```bash
rosbag play --clock drone.bag
```
```
rostopic echo /clock
```

```bash
---
clock: 
  secs: 1528107526
  nsecs: 869450146
---
clock: 
  secs: 1528116547
  nsecs: 629184961
---
clock: 
  secs: 1528116547
  nsecs: 634226202
---
clock: 
  secs: 1528107526
  nsecs: 879927647
---
clock: 
  secs: 1528116547
  nsecs: 639288617
---
clock: 
  secs: 1528116547
  nsecs: 644327139

```

droneboxid, 2018-06-14:
Hi - can you let me know how i can do with this data? please can you make some launch file for this? thanks

droneboxid, 2018-06-14:
i try to remap but nothing happen in GPS. it's same running without GPS. please can you advise me?

`<?xml version="1.0"?>
<launch>
  
  <arg name="nodelet_manager" default="velodyne_nodelet_manager" />
  <arg name="enable_floor_detection" default="true" />
  <arg name="enable_gps" default="true" />

  
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>

  
  <node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="load hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">
  	<remap from="/velodyne_points" to="/rslidar_points"/>
	<remap from="/mavros/global_position/global" to="/gps/geopoint"/> 
	<remap from="/mavros/global_position/raw/fix" to="/gps/fix"/> 
    <param name="use_distance_filter" value="true" />
    <param name="distance_near_thresh" value="1.0" />
    <param name="distance_far_thresh" value="50.0" />
    
    <param name="downsample_method" value="VOXELGRID" />
    <param name="downsample_resolution" value="0.2" />
    
    <param name="outlier_removal_method" value="RADIUS" />
    <param name="statistical_mean_k" value="30" />
    <param name="statistical_stddev" value="1.2" />
    <param name="radius_radius" value="0.5" />
    <param name="radius_min_neighbors" value="5" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="load hdl_graph_slam/ScanMatchingOdometryNodelet $(arg nodelet_manager)">
      <param name="keyframe_delta_trans" value="0.5" />
      <param name="keyframe_delta_angle" value="1.0" />
      <param name="keyframe_delta_time" value="2.0" />
      <param name="transform_thresholding" value="false" />
      <param name="max_acceptable_trans" value="2.0" />
      <param name="max_acceptable_angle" value="1.0" />
      <param name="downsample_method" value="NONE" />
      <param name="downsample_resolution" value="0.1" />
      
      <param name="registration_method" value="NDT_OMP" />
      <param name="ndt_resolution" value="2.0" />
      <param name="ndt_num_threads" value="0" />
      <param name="ndt_nn_search_method" value="DIRECT7" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="floor_detection_nodelet" args="load hdl_graph_slam/FloorDetectionNodelet $(arg nodelet_manager)"  if="$(arg enable_floor_detection)">
    <param name="sensor_height" value="2.0" />
    <param name="height_clip_range" value="0.5" />
    <param name="floor_pts_thresh" value="1024" />
    <param name="floor_normal_thresh" value="5.0" />
    <param name="use_normal_fitlering" value="true" />
    <param name="normal_filter_thresh" value="10.0" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="hdl_graph_slam_nodelet" args="load hdl_graph_slam/HdlGraphSlamNodelet $(arg nodelet_manager)">
    
    <param name="enable_gps" value="$(arg enable_gps)" />
    <param name="max_keyframes_per_update" value="10" />
    <param name="keyframe_delta_trans" value="0.1" />
    <param name="keyframe_delta_angle" value="3.0" />
    
    <param name="distance_thresh" value="30.0" />
    <param name="accum_distance_thresh" value="60.0" />
    <param name="min_edge_interval" value="10.0" />
    <param name="fitness_score_thresh" value="2.0" />
    
    <param name="registration_method" value="NDT_OMP" />
    <param name="ndt_resolution" value="2.0" />
    <param name="ndt_num_threads" value="0" />
    <param name="ndt_nn_search_method" value="DIRECT7" />
    
    <param name="floor_edge_stddev" value="100.0" />
    <param name="gps_edge_stddev" value="1000.0" />
    <param name="use_const_inf_matrix" value="false" />
    <param name="const_stddev_x" value="0.5" />
    <param name="const_stddev_q" value="0.1" />
    <param name="var_gain_a" value="20.0" />
    <param name="min_stddev_x" value="0.1" />
    <param name="max_stddev_x" value="5.0" />
    <param name="min_stddev_q" value="0.05" />
    <param name="max_stddev_q" value="0.2" />
    
    <param name="graph_update_interval" value="3.0" />
    <param name="map_cloud_update_interval" value="10.0" />
    <param name="map_cloud_resolution" value="0.05" />
  </node>

  <node pkg="hdl_graph_slam" type="map2odom_publisher.py" name="map2odom_publisher" />
</launch>`

droneboxid, 2018-06-14:
<remap from="/velodyne_points" to="/rslidar_points"/>
	<remap from="/mavros/global_position/global" to="/gps/geopoint"/> 
	<remap from="/mavros/global_position/raw/fix" to="/gps/fix"/>

koide3, 2018-06-15:
hdl_graph_slam accepts [geographic_msg/GeoPointStamped](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L222) or [nmea_msgs/Sentence](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L202). It seems your bag has another type of GPS data. Please convert your GPS data into either of them.

droneboxid, 2018-06-16:
please let me know how to convert that's from my BAG? thanks

droneboxid, 2018-06-16:
maybe if you want to make a donation or something like that, i can do that. very need to finish this project for my work. you can email me to pt.pumaeng@gmail.com as soon. thanks

closed 2018-08-13 by koide3