koide3/hdl_graph_slam #8: Failed to load nodelet hdl_graph_slam_nodelet

thomas3016, 2018-04-30:
Thank you for author greate job, I met a problem like
[ERROR] [1525081595.448920058, 1509878536.248604299]: Failed to load nodelet [/hdl_graph_slam_nodelet] of type [hdl_graph_slam/HdlGraphSlamNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type hdl_graph_slam::HdlGraphSlamNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1525081595.448999330, 1509878536.248604299]: The error before refreshing the cache was: Failed to load library /home/user/Autoware/ros/devel/lib//libhdl_graph_slam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/user/Autoware/ros/devel/lib//libhdl_graph_slam_nodelet.so: undefined symbol: _ZTIN3g2o16OptimizableGraph4EdgeE)
[FATAL] [1525081595.449254825, 1509878536.248604299]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `hdl_nodelet_manager'
[hdl_graph_slam_nodelet-5] process has died [pid 22601, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet hdl_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/user/.ros/log/2445a0dc-4c51-11e8-9319-0cc47a135e00/hdl_graph_slam_nodelet-5.log].
log file: /home/user/.ros/log/2445a0dc-4c51-11e8-9319-0cc47a135e00/hdl_graph_slam_nodelet-5*.log, can you give me some advice, I would be appreciated!

koide3, 2018-05-13:
Hi,
Would you try the way mentioned in Issue #4?

thomas3016, 2018-05-18:
@koide3 Thank you for  your mention, I have save this problem. I have been able to use the dat your have provided, but I had problem using my own data.  I used our platform to collect data of velodyne hdl-32e and save it as *.bag file. Then I start built map by running hdl_graph_slam_501.launch. I found two problem. Firstly,  the speed of mapping is not as fast as rosbag playing. When playing of rosbag is finished, the mapping is also stop. Secondly, when starting to build the map, you can build the map correctly, but when the map is over 100 m, the point cloud cannot be properly matched. Have you ever had a problem like this?

closed 2018-05-20 by thomas3016