koide3/hdl_graph_slam #10: Failed to load nodelet [/prefiltering_nodelet] of type [hdl_graph_slam/PrefilteringNodelet]

ekow-mensah, 2018-05-16:
Hi, I am trying to implement the hdl_graph_slam in real time using the Velodyne VLP-16 LiDAR, however I keep getting this error 

process[velodyne_nodelet_manager-1]: started with pid [3596]
process[prefiltering_nodelet-2]: started with pid [3597]
process[scan_matching_odometry_nodelet-3]: started with pid [3598]
process[floor_detection_nodelet-4]: started with pid [3605]
process[hdl_graph_slam_nodelet-5]: started with pid [3625]
process[map2odom_publisher-6]: started with pid [3641]
[FATAL] [1526472122.134142445]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[ INFO] [1526472122.136054231]: Initializing nodelet with 4 worker threads.
[FATAL] [1526472122.156214309]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[prefiltering_nodelet-2] process has died [pid 3597, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager __name:=prefiltering_nodelet __log:=/home/lunet/coem2/.ros/log/b1897200-5900-11e8-b12f-9c8e99dee3fc/prefiltering_nodelet-2.log].
log file: /home/lunet/coem2/.ros/log/b1897200-5900-11e8-b12f-9c8e99dee3fc/prefiltering_nodelet-2*.log
[scan_matching_odometry_nodelet-3] process has died [pid 3598, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/lunet/coem2/.ros/log/b1897200-5900-11e8-b12f-9c8e99dee3fc/scan_matching_odometry_nodelet-3.log].
log file: /home/lunet/coem2/.ros/log/b1897200-5900-11e8-b12f-9c8e99dee3fc/scan_matching_odometry_nodelet-3*.log

Some help will be appreciated.

koide3, 2018-05-16:
It seems, in your case, loading of the nodelet module has some problems. Please try the following:
1. Check if catkin_ws/devel/lib/libprefiltering_nodelet is correctly generated.
2. Re-build the package after removing catkin_ws/(devel|build)
3. Try the docker environment (see README)

koide3, 2018-05-16:
Hmmm, it suggests hdl_graph_slam_501 may have some problem. You see the error with other launch files? I'll check them.

ekow-mensah, 2018-05-20:
Sorry I gave you the wrong launch file it is hdl_graph_slam.launch that seems to generate that error. hdl_graph_slam_501 and 400 launch with no errors. I am really sorry about that.

closed 2018-08-13 by koide3