koide3/hdl_graph_slam #11: Using IMU with hdl_graph_slam

ekow-mensah, 2018-05-16:
Hi, Please I would like to ask if I was to implement the hdl_graph_slam algorithm with a mobile robot will it be possible to integrate an IMU to work with the algorithm or use odometry?

koide3, 2018-05-16:
For now, we cannot use IMU and odometry for hdl_graph_slam. It estimates the sensor motion from only range data. But, yes, we can improve the algorithm (in particular scan matching and graph optimization) with such information, and I want to do it someday.

The launch files are tuned for different environments (400: indoor, 500: outdoor, ford: outdoor with GPS). You can see example environments in README.

ekow-mensah, 2018-05-16:
Ahhh I see thank you. Do you reckon including IMU data and odometry will be challenging to implement?

koide3, 2018-05-16:
No, theoretically, it is not so difficult, and I already have some code for the integration of IMU and scan matching for another project (hdl_graph_slam). Just time is needed...

ekow-mensah, 2018-05-16:
Ahhh okay thank you very much for the help I really appreciate it.

closed 2018-05-16 by ekow-mensah

weisongwen, 2019-07-06:
hi @koide3 , thanks for sharing your work. The IMU acceleration is integrated into the graph optimization. however, the bias for IMU is not estimated yet?

GLKode, 2019-10-01:
Hi @koide3 , @weisongwen @ekow-mensah 
hdl_graph_slam is mainly using odometry data (in keyframes, scan matching, ..), I would like to use the IMU instead of the odometer data, or maybe both together, could you please give me some hints how this is possible and which classes, nodelets to be changed? do we need to implement IMU wherever odometer is used? 
Thank you.