koide3/hdl_graph_slam #12: Mapping faild when used our own data

thomas3016, 2018-05-20:
I have been able to use the data your have provided, but I had problem using my own data. I used our platform to collect data of velodyne hdl-32e and save it as *.bag file. Then I start built map by running hdl_graph_slam_501.launch. I found two problem. Firstly, the speed of mapping is not as fast as rosbag playing. When playing of rosbag is finished, the mapping is also stop. Secondly, when starting to build the map, you can build the map correctly, but when the map is over 100 m, the point cloud cannot be properly matched. Have you ever had a problem like this?

koide3, 2018-05-20:
1. The mapping speed depends on your PC. If it is too slow, try *bag_player.py* which adjusts the playback speed automatically. 

2. You mean the scan matching was failed? If the sensor moved very quickly, it might a reason.

thomas3016, 2018-06-24:
Thank you for your answer. I tested your suggestion. First, use the bag_player.py to play the *.bag offline at idle speed. We cannot get the complete result. Then build the map online in real time, still can't get the complete result. If it is a problem with the configuration of my computer, can you tell me what kind of configuration I need  ?

koide3, 2018-06-25:
Hi @thomas3016 ,
I cannot figure out what is happing on your PC. Can you upload a movie or screenshots of the mapping?

thomas3016, 2018-06-28:
Hi @koide3, when I played *.bag at a very low speed, I finally succeeded in building the map. So the reason I didn't build it before was mainly because the real-time performance of my computer could not be achieved. A new problem that arose along the way was that I did not use GPS and inertial navigation, and the figure built on the outdoor test site was seriously deformed. Especially after passing the curve, did you encounter such a problem?

thomas3016, 2018-07-24:
Thanks for your help@koide3

closed 2018-07-24 by thomas3016