koide3/hdl_graph_slam #13: Failed to load nodelet

aaravrav142, 2018-05-29:
Hi,

I have installed the package correctly on ROS kinetic following the guide. g2o was also installed without error. I also have velodyne drivers sourced installed inside the catkin_ws. When I try to launch any file through the package, I am getting the following errors. I am not sure why it is not able to find the velodyne nodelet manager.  Any suggestions would be helpful. Thanks

```
[FATAL] [1527575498.004853517]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1527575498.004870101]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1527575498.004927881]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'

```

hunkyu, 2018-06-29:
[FATAL] [1530286929.069876848]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1530286929.069878517]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1530286929.069876867]: Failed to load nodelet '/floor_detection_nodelet` of type `hdl_graph_slam/FloorDetectionNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1530286929.069892040]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-1] process has died [pid 25697, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/velodyne_nodelet_manager-1.log].
log file: /home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/velodyne_nodelet_manager-1*.log
[prefiltering_nodelet-2] process has died [pid 25698, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager __name:=prefiltering_nodelet __log:=/home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/prefiltering_nodelet-2.log].
log file: /home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/prefiltering_nodelet-2*.log
[scan_matching_odometry_nodelet-3] process has died [pid 25699, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/scan_matching_odometry_nodelet-3.log].
log file: /home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/scan_matching_odometry_nodelet-3*.log
[floor_detection_nodelet-4] process has died [pid 25701, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/FloorDetectionNodelet velodyne_nodelet_manager __name:=floor_detection_nodelet __log:=/home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/floor_detection_nodelet-4.log].
log file: /home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/floor_detection_nodelet-4*.log
[hdl_graph_slam_nodelet-5] process has died [pid 25714, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/hdl_graph_slam_nodelet-5.log].
log file: /home/hunkyu/.ros/log/a51b59ae-7bae-11e8-bf16-e0d55ea7d733/hdl_graph_slam_nodelet-5*.log

aaravrav142, 2018-07-02:
Hi,

Issue #4 did not solved the problem. I changed the system and the problem persists. I am using g2o version as suggested in the readme. If there are any updates please share.

closed 2018-08-13 by koide3

Gaochao-hit, 2018-08-27:
Try this method, it helpps me:   https://github.com/koide3/hdl_graph_slam/issues/4#issuecomment-410645166

YangSiri, 2019-12-13:
Hi, @hunkyu 

Did you find any solution to your problem?