koide3/hdl_graph_slam #15: pose of final graph optimization

weisongwen, 2018-06-12:
hi @koide3 

Thanks for sharing your great work here,
The topic /odom contains the pose of scan_matching (without the graph optimization). But how can we obtain the pose from the final graph optimization of each epoch? 

best 
welson

weisongwen, 2018-06-13:
hi @koide3 ,
thanks for you kind reply. actually, I want to evaluate the performance of the mapping process (the optimized pose)? Could you please suggest how to obtain the optimizaed pose?
Best 
Welson

koide3, 2018-06-13:
hi @weisongwen ,
OK, you can use */hdl_graph_slam/dump* service to obtain the optimized graph. It saves all the graph data, and you can take the graph file (graph.g2o) to obtain the optimized pose of each node.

```bash
$:~/.ros/13-06-2018 08:57:22$ ls
000000  000001  000002  000003  000004  graph.g2o

$:~/.ros/13-06-2018 08:57:22$ head -n 5 graph.g2o
VERTEX_PLANE 0 0 0 1 0 0.2 0.2 0.2
FIX 0
VERTEX_SE3:QUAT 1 0 0 0 0 0 0 1
VERTEX_SE3:QUAT 2 0.0820318 0.522197 0.00859015 -0.115912 0.0276225 0.0217396 0.992637
VERTEX_SE3:QUAT 3 -0.115682 1.07558 -0.0260632 -0.0019982 0.00531704 -0.00327759 0.999978

$:~/.ros/13-06-2018 08:57:22$ ls 000004/
cloud.pcd  data

$:~/.ros/13-06-2018 08:57:22$ cat 000004/data
stamp 1509348800 463194000
odom
   0.985232   -0.171065 -0.00743227   -0.438065
   0.171054    0.981373   0.0874442     2.06066
-0.00766478  -0.0874242    0.996142  -0.0277345
          0           0           0           1
accum_distance 2.15623
floor_coeffs 0.0317455 -0.074796  0.996693   1.81636
id 5
```

weisongwen, 2018-06-14:
Dear @koide3 ,
Thanks for your kind information, this is a good idea to evaluate the performance of the map generation. However, I want to evaluate the real-time performance of this graph SLAM. 
I want to obtain the real time optimized pose. what you suggested is the final graph after the whole mapping process.
[image]
According to your flowchart, the **/odom** is the pose fully from scan_matching between **keyframe** and **keyframe** without the **conpensation from graph optimization**. What i needed is to obtain the real time pose (as a ros topic) from the optimization.
best 
welson

koide3, 2018-06-15:
@weisongwen ,
In the current implementation, it doesn't output any real-time optimization results. You need to modify the code a bit. You can save the optimized graph with;

```
std::lock_guard<std::mutex> lock(main_thread_mutex);
graph_slam->save("output_filename.g2o");
```
See [the code of the dump service](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L619). Please put the code at some place you want to obtain the graph. I think [just after the optimization](https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L415) is a good place to put the code.

weisongwen, 2018-06-17:
Dear @koide3 ,
thanks for your kind information,

best 
welson

closed 2018-06-19 by koide3

mnik17, 2019-07-14:
@weisongwen 
Hi,
I'm trying to do the same thing, so I just wanted to ask if you managed to do it and how?

Thanks in advance

weisongwen, 2019-07-15:
@mnik17 , you can get it from here.  https://github.com/koide3/hdl_graph_slam/blob/e7b5433f467e946579521817b3d24bb2cc208b95/apps/hdl_graph_slam_nodelet.cpp#L174

ChristopherLu, 2020-01-31:
odom2map * keyframe->odom; 

Is it the sensor pose after graph optimization or it is simply the result after scan matching?

koide3, 2020-02-01:
Hi @ChristopherLu ,
``keyframe->odom`` is a result of the scan matching without graph optimization, and ``odom2map`` is a transformation which corrects the odometry error and aligns the (latest) odometry frame with respect to the map frame. So, ``odom2map * keyframe->odom`` is the keyframe pose in the map frame after graph optimization (Note that it can jump).

wongkenchiang97, 2021-08-28:
Hi, thanks for sharing a creative work!! I'm also working to get the optimized trajectory. 

I have collected odom2map(topic: /odom2pub ) and keyframe->odom(topic: /tf)

The length of odom2map transform and keyframe->odom pose are different. 

Any advise for getting corresponding odom2map and keyframe->odom?