koide3/hdl_graph_slam #16: error in rviz

weisongwen, 2018-06-14:
hi @koide3  
when I run my own bag file, the tf transform between the odom and the keyframe/velodyne seems to have not been published. do you have any suggestion on this?  
[image]

best 
welson

koide3, 2018-06-15:
@weisongwen ,
It sometimes happens when the program couldn't process data real-time. Please try to use bag_player.py or play your bag at a slow speed.

weisongwen, 2018-06-17:
Dear @koide3 ,
thanks for your information,
best 
welson

closed 2018-06-19 by koide3

caoyifeng001, 2019-05-23:
@weisongwen hello how to solve it ,i have same problem.

weisongwen, 2019-05-23:
as @koide3  mentioned, play your bag at a slow spped.