koide3/hdl_graph_slam #17: pose graph optimization  error

autoperson, 2018-06-20:
Hi ,I already install g2o with hdl_graph_slam branch ,but when pose graph optimization , it has problem with [velodyne_nodelet_manager-2] process has died [pid 4467, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager , am I need set g2o dir in cmakelist?

autoperson, 2018-06-20:
and can I get your paper?  please send to autoperson@qq.com

mameng1, 2018-06-25:
I also need your paper,can you send your paper to 2908071445@qq.com

sysuequal, 2018-06-26:
I have the same problem that the [velodyne_nodelet_manager-1] process has died

koide3, 2018-06-29:
Did you try the ways in #4 ? In particular, try docker.

Regarding the paper, I think I can send you it if you tell me your official address in your institute or organization.

mameng1, 2018-06-29:
@koide3,hello,I am a student in xidian' university ,Xian,China,my email is qmma@stu.xidian.edu.cn

mameng1, 2018-06-29:
@koide3,hello,I am a student in xidian' university ,Xian,China,my email is qmma@stu.xidian.edu.cn,thank you

autoperson, 2018-07-04:
I don't find the problem again , may be I adjust optimize iterations param

closed 2018-07-04 by autoperson

autoperson, 2018-07-06:
I set the iteration param with 512

koide3, 2018-07-06:
@autoperson ,
Thanks for the information. I'll try the setting...

ekow-mensah, 2018-08-06:
@autoperson Where did you set the iteration param?? In the launch files? or the actual code?

autoperson, 2018-08-07:
It was the number of  g2o optimization iterations in code @ekow-mensah, you can find other answers of 4# question.