koide3/hdl_graph_slam #19: Failed to use my own .bag data for mapping

sunmiaozju, 2018-06-25:
Hi, we can run the example *.bag data to build the map correctly, but there is something wrong when i am using my own *.bag data. (i use a velodyne-vlp16). Here are the details:

i use the command to record my own *.bag:

`roslaunch velodyne_pointcloud VLP16_points.launch [calibration:=/home/smartcar/sunm/velodyne/VeloView-VLP-16-HiRes.yaml] `

`rosbag record -O indoor /velodyne_points `

does this is right to save my own *.bag data ? thanks a lot

when i using my own  *.bag data , i got the error message in the rviz:
[image]

and, i got the error message ine the terminal:

[image]

and in the rviz, i only get the motionless point, it seems like it only show one frame:

[image]

it seems that there is something wrong in the time of the data recorded according to the error mssage in the rviz. But i don't know how to deal with it ? Would help me with it? thanks a lot!

koide3, 2018-06-26:
Hi @sunmiaozju ,
When you played your bag, did you added --clock option? like:
```bash
rosbag play --clock ***.bag
```
Without this option, tf cannot handle timestamps in a bag properly.

sunmiaozju, 2018-06-27:
Thank you for your answer.  But i indeed use the --clock option to output the timestamps. 

I think if the equipment (i am using velodyne-vlp16) is not suitable for the code, because i find that the example bag is from velodyne HDL32e.

can we only use the velodyne-vlp16 for the code? Thanks a lot!

koide3, 2018-06-29:
Hi @sunmiaozju ,
We've tested this package with VLP16 as well. But, it could be unstable compared to HLD32e, since it has a narrow field of view and provide fewer points. I think you have to move the sensor slowly.

To check if the scan matching is working, change "Fixed Frame" on rviz to "odom". You'll be able to see the odometry estimated by the matching.

closed 2018-08-13 by koide3