koide3/hdl_graph_slam #21: Failed to run the order for saving map

ikunjin, 2018-07-19:
Hi,the following error occurs when I run "rosservice call /hdl_graph_slam/save_map "resolution: 0.05
destination: '/full_path_directory/map.pcd'"" of example 1 in my own PC.

ERROR: service [/hdl_graph_slam/save_map] responded with an error: : [pcl::PCDWriter::writeBinary] Error during open!

Please can you help me figure out this problem?
Thanks a lot!

koide3, 2018-07-26:
I think you have to specify a destination where the map will be written instead of "/full_path_directory/map.pcd" (e.g., "/tmp/map.pcd").

weisongwen, 2018-08-07:
hi @koide3 , the command rosservice call /hdl_graph_slam/save_map "resolution: 0.05
destination: '/full_path_directory/map.pcd'" did not work. For example, i want to save the map into my computer with the directory "/home/wenws/mapfile.pcd", the resolution is 0.1. what is the full command to do this?

Best
welson

koide3, 2018-08-13:
hi @weisongwen ,
The command should be:
```bash
rosservice call /hdl_graph_slam/save_map "resolution: 0.1
destination: '/home/wenws/mapfile.pcd'"
```
How the command failed? Nothing happened or the program got crashed?

ikunjin, 2018-08-27:
Hi,@koide3 ,now it works well.
Thank you very much!!

ikunjin, 2018-09-19:
Hi,@koide3,it works well with the .bag data collected by my own lidar.But when i do real-time slam with my lidar and  also remap the topic,it fails.It seems that only the .bag file works.
Please could you tell me what i should do to deal with the problem?
Thanks a lot!!

koide3, 2018-09-24:
Hi @ikunjin,
How did it fail? Please check if use_sim_time is set to false.

ikunjin, 2018-09-25:
Hi,@koide3,thank you for your advice.When I set use_sim_time to false,the problem has disappeared. 
Thank you very much !

tranducthuan1, 2018-10-03:
Dear Prof. Koide,

I have a problem when saving the map file.

The error shows that: Unable to load type [hdl_graph_slam/SaveMap].
then, : have you typed 'make' in [hdl_graph_slam]?

Could you help me to solve it, please.

I  look forward to hearing from you.

Yours sincerely,
Tran

tranducthuan1, 2018-10-03:
after I changed the "source devel/setup.bash" .
The error  is yaml.scanner.ScannerError: mapping values are not allowed here

BurryChen, 2018-10-06:
I have also the error:
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<string>", line 1, column 29:
    resolution: 0.05 destination: '/home/whu/data/hdl_graph/map.pcd'

BurryChen, 2018-10-08:
@tranducthuan1 
I have solved the problem.
After `rviz -d hdl_graph_slam.rviz`, then execute two lines command
`rosservice call /hdl_graph_slam/save_map "resolution: 0.05
destination: '/full_path_directory/map.pcd'"` with "\".

tranducthuan1, 2018-10-08:
Dear @BurryChen ,

Thank you so much.

Actually, I also solved it problem yesterday.

I am sorry for my late response.

Did you try this map for the next step of localization?

I have got an error of the tf, the map has not yet match the localization.

Tran

koide3, 2018-10-22:
hi @weisongwen ,
I found that the code didn't store the data for map generation when no one subscribes the map topic for visualization. Could you try the latest commit in which the bug is fixed?

Best,
Kenji

improve100, 2018-10-23:
> @tranducthuan1
> I have solved the problem.
> After `rviz -d hdl_graph_slam.rviz`, then execute two lines command
> `rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/full_path_directory/map.pcd'"` with "\".

i try 
$ rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/full_path_directory/map.pcd'" failed
but this is ok.
$ rosservice call /hdl_graph_slam/save_map "resolution: 0.05 </br>
  destination: '/full_path_directory/map.pcd'"

koide3, 2018-10-23:
@ikunjin 
This is a known bug on PCL 1.7 binary. To avoid it, you have to disable c++11 option or re-build pcl from the source. In short, it's so hard to get rid of this bug.

https://github.com/PointCloudLibrary/pcl/issues/619
https://stackoverflow.com/questions/26346690/debug-seg-fault-in-boostmathlanczos-with-libpcl-surface

koide3, 2018-10-23:
@tranducthuan1 
It's just a format error. Use auto-completion by pressing TAB, and do not copy and paste the command.

```bash
rosservice call /hdl_graph_slam/save_map <TAB><TAB>
```

You'll see a blank data in the correct format like this. Just fill the arguments.

```bash
rosservice call /hdl_graph_slam/save_map "utm: false
resolution: 0.0
destination: ''" 
```

ikunjin, 2018-10-24:
Hi, @koide3, thanks very much for your guidance, i solve it by rebuilding pcl from the source. But now i have a new problem. I add "$ENV{CXXFLAGS} -O0 -Wall -g -ggdb"  to CMAKE_CXX_FLAGS and set CMAKE_BUILD_TYPE to Debug in the CmakeLists.txt, also adding "launch-prefix="xterm -e gdb --args" " to the .launch file(for example, the prefiltering_nodelet). The xterm tells  "no debug symbols found" and "no executable file specified". It seems that using nodelet, different from node,cmake generates dynamic link library--the .so file by the CmakeLists.txt, so gdb can not work. Please can you help me point out where i am wrong?
Thank you very much again ! !

koide3, 2018-10-25:
hi @ikunjin ,
Unfortunately, I'm not familiar with running a debugger on nodelets. I think the following topic would be helpful for you.
http://ros-users.122217.n3.nabble.com/debugging-nodelets-td2370662.html

ikunjin, 2018-10-26:
Hi @koide3,I will continue to learn about this issue.Anyway,i am extremely grateful for your precious advice.

DavidJamesOKC, 2018-12-08:
hello, @koide3 , I want to save a map.pcd file by running "rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/data/map.pcd'"". But the terminal show a error:
ERROR: Service [/hdl_graph_slam/save_map] is not available.
I try solve, but it still is this error.
I'm looking forward your reply. thanks.

imawesome61, 2018-12-18:
Respected sir's, i'm trying to do this on ubuntu 18.04 and none of it works can please any of you help me and tell me which ros should i install and how will i execute this code thanking you in advance .

koide3, 2018-12-19:
hi @DavidJamesOKC ,
Could you try "rosnode list" to check if the nodelets are launched correctly?

sanjanawill, 2019-03-20:
> @tranducthuan1
> I have solved the problem.
> After `rviz -d hdl_graph_slam.rviz`, then execute two lines command
> `rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/full_path_directory/map.pcd'"` with "\".

> @tranducthuan1
> I have solved the problem.
> After `rviz -d hdl_graph_slam.rviz`, then execute two lines command
> `rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/full_path_directory/map.pcd'"` with "\".

Hi 

> @tranducthuan1
> I have solved the problem.
> After `rviz -d hdl_graph_slam.rviz`, then execute two lines command
> `rosservice call /hdl_graph_slam/save_map "resolution: 0.05 destination: '/full_path_directory/map.pcd'"` with "\".

Hi, @BurryChen 
I got the same error.  Could you please specify the command in more detail.

Thank you.

Leedk3, 2019-09-14:
@DavidJamesOKC Did you solve the problem? I'm suffering the same issue now.
I can't see the "/save_map" service in the ros-service list.

/hdl_graph_slam_nodelet/get_loggers
/hdl_graph_slam_nodelet/set_logger_level
/lidar2base_publisher/get_loggers
/lidar2base_publisher/set_logger_level
/map2odom_publisher/get_loggers
/map2odom_publisher/set_logger_level
/prefiltering_nodelet/get_loggers
/prefiltering_nodelet/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/scan_matching_odometry_nodelet/get_loggers
/scan_matching_odometry_nodelet/set_logger_level

Leedk3, 2019-09-16:
@koide3 
Hello. I have a question. 
When I enter the command as

rosservice call /hdl_graph_slam/save_map "utm: false
resolution: 0.1
destination: '/home/lee/catkin_ws/src/hdl_graph_slam/map/map.pcd'" 

the return was " success: False "

I think the rosservice command is same with the others. 

Can you give me any comment?

Thank you!

Best, 
Lee

NohSeungWook, 2024-01-22:
I solved this problem by entering the command below.

rosservice call /hdl_graph_slam/save_map false 0.05 '/you_full_path_directory/map.pcd'

Yes, I think this is a simple issue of unfamiliarity with rosservice - if you only enter the argument as a list of values, not in property: value format, it will usually work.