koide3/hdl_graph_slam #22: Algorithm crashing

ekow-mensah, 2018-08-03:
Hi, 
I have been experiencing a problem where after a number of nodes are added to the pose graph the algorithm crashes and returns the following errors. 

[pcl::NormalEstimation::compute] input_ is empty!
 
[ WARN] [1533303347.367675061]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty

I would really appreciate some help on this.

koide3, 2018-08-07:
Hi,
I guess the first error was caused in floor_detection_nodelet since no other nodelet uses the normal estimation. Can you disable the floor detection by changing "enable_floor_detection" in your launch file to "false"?

ekow-mensah, 2018-08-07:
When i set the floor_detection_nodelet in the launch file to false, it still crashes and returns a long error message. Please see below: 

[ERROR] [1533638036.111240000]: Failed to load nodelet [/prefiltering_nodelet] of type [hdl_graph_slam/PrefilteringNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class hdl_graph_slam/PrefilteringNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet
[ERROR] [1533638036.111341937]: The error before refreshing the cache was: According to the loaded plugin descriptions the class hdl_graph_slam/PrefilteringNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet
[ERROR] [1533638036.133664307]: Failed to load nodelet [/scan_matching_odometry_nodelet] of type [hdl_graph_slam/ScanMatchingOdometryNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class hdl_graph_slam/ScanMatchingOdometryNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne [... cut ...] Synchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet

koide3, 2018-08-07:
Hi,
Can you start all the nodelets with standalone mode to find which one gets crashed? 

Change:
<node pkg="nodelet" ... args="load ..." />
to:
<node pkg="nodelet" ... args="standalone ..." />

ekow-mensah, 2018-08-07:
hi, when i started all nodelets in standalone mode, i found that the hdl_graph_slam nodelet keeps crashing. I have pasted the error below: 

[hdl_graph_slam_nodelet-3] process has died [pid 16590, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/lunet/coem2/.ros/log/e1bbd4f2-9a39-11e8-ae46-9c8e99dee3fc/hdl_graph_slam_nodelet-3.log].
log file: /home/lunet/coem2/.ros/log/e1bbd4f2-9a39-11e8-ae46-9c8e99dee3fc/hdl_graph_slam_nodelet-3*.log

koide3, 2018-08-07:
hi,
Are you using the branch "devel"? It seems the optimizer causes the error. Can you change "g2o_solver_type" to "lm_var" or "gn_var"?

ekow-mensah, 2018-08-08:
hi, 
I am using the "lm_var" solver and no I forked the project from the master branch. Should I switch the solver type to "gn_var"?

koide3, 2018-08-13:
I think floor constraints may significantly improve the map accuracy in your environment. Try to enable floor_detection_nodelet and check if the floor is correctly detected by looking /floor_detection/floor_filtered_points.

ekow-mensah, 2018-08-13:
Thanks @koide3  I will try doing that and let you know how it goes.

closed 2018-08-26 by ekow-mensah