koide3/hdl_graph_slam #23: crash when optimizing

planckwu, 2018-08-10:
1.testing Example1 (Indoor)
2.g2o:https://github.com/koide3/g2o.git

crash when optimizing:

ROS_MASTER_URI=http://localhost:11311

process[nodelet_manager-1]: started with pid [2710]
process[prefiltering_nodelet-2]: started with pid [2711]
process[scan_matching_odometry_nodelet-3]: started with pid [2712]
process[floor_detection_nodelet-4]: started with pid [2713]
process[hdl_graph_slam_nodelet-5]: started with pid [2714]
process[map2odom_publisher-6]: started with pid [2715]
[ INFO] [1533869775.044699738]: Initializing nodelet with 8 worker threads.
downsample: NONE
registration: NDT_OMP DIRECT7 1 (0 threads)
downsample: VOXELGRID 0.1
outlier_removal: RADIUS 0.5 - 2
construct solver... 
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
done
registration: NDT_OMP DIRECT7 1 (0 threads)

--- pose graph optimization ---
nodes: 7   edges: 10
optimizing... [nodelet_manager-1] process has died [pid 2710, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/nodelet_manager-1.log].
log file: /home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/nodelet_manager-1*.log
[scan_matching_odometry_nodelet-3] process has finished cleanly
log file: /home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/scan_matching_odometry_nodelet-3*.log
[prefiltering_nodelet-2] process has finished cleanly
log file: /home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/prefiltering_nodelet-2*.log
[floor_detection_nodelet-4] process has finished cleanly
log file: /home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/floor_detection_nodelet-4*.log
[hdl_graph_slam_nodelet-5] process has finished cleanly
log file: /home/haylion/.ros/log/d05bb482-9c48-11e8-947a-2cfda189a91d/hdl_graph_slam_nodelet-5*.log

zyyfly2010, 2018-08-10:
@planckwu same problem as #4 , you cam try my way by change g2o lib

planckwu, 2018-08-11:
@zyyfly2010 Thank you so much! I had tried as you said and it solved the problem perfectly.

closed 2018-08-13 by koide3