koide3/hdl_graph_slam #25: Map odom transforms and mapping an unflat surface

ashammaa, 2018-08-16:
Hello,

I have installed you Hdl_graph_slam and Hdl_localizations packages and have been implementing them on a Clearpath Husky robot and have a couple of questions:

Hdl_graph_slam:
1- I recorded a rosbag file of the area I want to map and launched the hdl_graph_slam launch file, however, during the mapping process, the odom frame jumps around, while only the map frame stays fixed to where the mapping process star

koide3, 2018-08-17:
Hi @ashammaa ,

hdl_graph_slam: 
1. It's correct behavior. odom frame is the origin of the scan matching "odometry". So, it can drift from the global origin (map frame) like usual wheel odometry.

2. Sorry for the late reply. I'll send you the paper.

3. I think you can obtain good results with GPS, and if your system has an IMU, please consider to use the "devel" branch in which I implemented new IMU-based constraints. They can improve the quality of the map in large environments. Regarding parameter tuning, "ndt_resolution" have a big impact on mapping accuracy. Please try several resolutions (0.5 ~ 2.0m in indoor, 1.0 ~ 5.0m in outdoor) for a good mapping result. 

hdl_localization:
I think you have to modify the tf tree so that map will be the root of the tree like [this figure](http://roboticsknowledgebase.com/wiki/state-estimation/assets/ROSNavigation-d06e2.png). Unfortunetely, in the current implementation, hdl_localization publishes transform between "map" and "velodyne", and the frames are hard coded [here](https://github.com/koide3/hdl_localization/blob/a6d4c4d649aecfe94f0bcf90a7d8216f323dae67/apps/hdl_localization_nodelet.cpp#L233). I expect that it works on your system if you rewrite the code so that it publishes transfrom between "map" and "base_link" properly.

Best,

ashammaa, 2018-09-03:
Hello @koide3,

Thank you for your reply! I have several more questions regarding the package:

Hdl_graph_slam:

1- Is the only difference between mapping from the lidar frame and the base_link the initialization of (0,0,0) of the map? In other words, if we are mapping from the lidar frame, the (0,0,0) of the map will be where the lidar is and if we are mapping from the base_link frame, the (0,0,0) of the map will be where the base_link frame is? 
2- What do the "---_edge_robust_kernel" parameters do for each of the gps and imu?
3- For clarification purposes, the "..._edge_stddev" specifies how much do we trust the measurement of that specific sensor, right?

Hdl_localization:

1- What does the "invert_imu" parameter do? In which way does it rotate the imu?

Regards,

Abed

koide3, 2018-09-04:
Hi @ashammaa ,

hdl_graph_slam:
1. Yes, the initial position of the lidar will be the origin of the map. Note that any map optimization (loop closure, GPS, IMU) may move the origin from the initial position.
2. Those parameters allow you to apply robust kernels (like Huber kernel) to the error calculation. They prevent map deformation due to wrong inputs (i.e., outliers) like wrongly matched loops. You can use any robust kernels supported by g2o.
3. It is used to calculate the information matrix of each observation. The information matrix is given by "information = identity_matrix / stddev". So, smaller stddev means more reliable observation.

hdl_localization:
1. If you enable it, angular velocity inputs are multiplied by "-1" (in other words, all the axes are inverted). I made this option for just compatibility with a specific LIDAR (HDL32e). For other sensors, maybe you don't have to use this option.

ashammaa, 2018-09-04:
Hello @koide3,

When I am using the hdl_localization package, for the motion models used in the prediction step of the Unscented Kalman Filter, both models (process noise or IMU) does not lead to the converging of the UKF. I believe that is because my Clearpath Husky robot has a different motion model and integrating the acceleration using the IMU is proven to be too noisy. For such a case, the only way to improve the localization would be to modify the code to fuse the information of the wheel odometry? Or is there another way to improve localization as the package is currently constructed?   

Thanks again for your time!!! 

Regards, 

Abed

koide3, 2018-09-05:
Hi @ashammaa ,

Yes, fusing the wheel odometry and the IMU data would be the best way. But, if you don't want to do that, another ad-hoc solution is to tune scan matching parameters. In particular, ndt_resolution has a big impact on the convergence of the localization. Try a bigger resolution (2~10m, depending on your LIDAR and environment).

closed 2018-09-12 by ashammaa