koide3/hdl_graph_slam #28: Hello,your work impresses me deeply,but I can't find the corresponding paper online.Could you please offer the manuscript  ''Kenji Koide, Jun Miura, and Emanuele Menegatti, A Portable 3D LIDAR-based System for Long-term and Wide-area People Behavior Measurement, IEEE Transactions on Human-Machine Systems (under review) '?

Gaochao-hit, 2018-08-27:


Gaochao-hit, 2018-08-29:
Hello,your work impresses me deeply,but I can't find the corresponding paper online.Could you please offer the manuscript  ''Kenji Koide, Jun Miura, and Emanuele Menegatti, A Portable 3D LIDAR-based System for Long-term and Wide-area People Behavior Measurement, IEEE Transactions on Human-Machine Systems (under review) '?

chennuo0125, 2018-08-30:
how can I get the paper? @koide3

koide3, 2018-08-30:
Hi,
In case you need the paper, send a mail to koide@aisl.cs.tut.ac.jp from your official address in your institute or organization. #6

2amour, 2018-09-02:
hi, @koide3 I sent you a mail to koide@aisl.cs.tut.ac.jp for the paper ,but you don't reply to me.

Gaochao-hit, 2018-09-21:
@koide3  hello,I have emailed to you using my enterprise mailbox,but have not received a reply.