koide3/hdl_graph_slam #29: map moves instead of velodyne tf marker

ekow-mensah, 2018-08-28:
hi @koide3 please I have a little problem. I have the VLP-16 lidar mounted on a tripod. As I move the tripod, the map depicted by the /velodyne_points topic moves along with the map while the "velodyne" tf marker remains stationary. Do you have any idea why this could be happening?

koide3, 2018-08-30:
@ekow-mensah ,
Can you check if the fixed frame of rviz is set to "map"?

ekow-mensah, 2018-08-30:
Yes It is set to map on rviz but it still moves around

koide3, 2018-09-04:
hi @ekow-mensah ,
How about "odom" frame?

ekow-mensah, 2018-09-04:
hi @koide3 the "odom" frame does not move at all It remains stationary.

koide3, 2018-09-05:
So, "map" frame moves while "velodyne" and "odom" frames remain stationary? It sounds very strange... Did you see such a problem on the example bag files and the rviz config?

ekow-mensah, 2018-09-05:
No this problem does not exist in the example bag files or rviz config. I must have done something wrong but I can't figure out what.

zcgzcg, 2018-12-29:
Hi, ekow-mensah , I had the same problem, I use HDL-32E, But I can not know  the reason for map moves, Can you give me some help. @ekow-mensah ,thanks !