koide3/hdl_graph_slam #31: complain about "fail to find transform"

tuandle, 2018-09-07:
Hi @koide3 ,
When I turn on `enable_imu_acc`, I notice that hdl_graph_slam can't find the transform between keyframes. I tried different `ndt_nn_search_method` as well as `kernel` types but they don't help with the problem. 
Could you give any hint that might cause this problem? My guess is about the time-sync between imu readings and velodyne scans. Is it better if they are synced?

koide3, 2018-09-07:
hi @tuandle ,

In the new version in devel branch, I modified the code so that all input data is transformed into "base_link" frame before processing since the lidar, imu, and gps may have different frames. To use IMU constraints, you have to describe the transform between the imu and base_link frames.

Sorry, I think I didn't document that, and the example launch file contains only lidar-base transformation publisher. In case the lidar and the imu share the same frame (like in the example bag), you can add following lines in the launch file:

```bash
  <node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 0 0 0 0 base_link velodyne 10" />
  <node pkg="tf" type="static_transform_publisher" name="imu2base_publisher" args="0 0 0 0 0 0 base_link imu_frame 10" />
```

tuandle, 2018-09-07:
@koide3 thank you so much for your fast reply! I should have dig deeper into your code :)

closed 2018-09-07 by tuandle