koide3/hdl_graph_slam #32: the prefilering_nodelet.cpp

2amour, 2018-09-08:
hi, @koide3 ,I want to ask baout the question about points_sub = nh.subscribe("/velodyne_points", 64, &PrefilteringNodelet::cloud_callback, this);in the prefiltering_nodelet.cpp.
void cloud_callback(const pcl::PointCloud<PointT>::ConstPtr& src_cloud); the type of the parameter is sensor_msgs/PointCloud2, but in your code is pcl::PointCloud<PointT>, and I could not understand it.
Thanks!

koide3, 2018-09-09:
hi @2amour ,
You can directly receive point cloud messages as pcl::PointCloud with pcl_ros (it automatically converts ros cloud to pcl cloud). See [here](http://wiki.ros.org/pcl_ros#Subscribing_to_point_clouds).

2amour, 2018-09-09:
Thanks a lot! @koide3

closed 2018-09-09 by 2amour