koide3/hdl_graph_slam #36: roslaunch hdl_graph_slam hdl_graph_slam.launch error

improve100, 2018-10-22:
[image]
process[lidar2base_publisher-1]: started with pid [1290]
process[velodyne_laser1-2]: started with pid [1291]
process[velodyne_nodelet_manager-3]: started with pid [1298]
process[prefiltering_nodelet-4]: started with pid [1315]
process[scan_matching_odometry_nodelet-5]: started with pid [1328]
process[hdl_graph_slam_nodelet-6]: started with pid [1330]
process[map2odom_publisher-7]: started with pid [1350]
[ INFO] [1540203422.088669992]: Initializing nodelet with 12 worker threads.
downsample: VOXELGRID 0.1
outlier_removal: RADIUS 0.5 - 2
downsample: NONE
registration: NDT_OMP DIRECT7 1 (0 threads)
construct solver...
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
registration: NDT_OMP DIRECT7 1 (0 threads)
[velodyne_nodelet_manager-3] process has died [pid 1298, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/tong/.ros/log/985a32fa-d5e3-11e8-a625-80ce62f28ef3/velodyne_nodelet_manager-3.log].
log file: /home/tong/.ros/log/985a32fa-d5e3-11e8-a625-80ce62f28ef3/velodyne_nodelet_manager-3*.log
[prefiltering_nodelet-4] process has finished cleanly
log file: /home/tong/.ros/log/985a32fa-d5e3-11e8-a625-80ce62f28ef3/prefiltering_nodelet-4*.log
[scan_matching_odometry_nodelet-5] process has finished cleanly
log file: /home/tong/.ros/log/985a32fa-d5e3-11e8-a625-80ce62f28ef3/scan_matching_odometry_nodelet-5*.log
[hdl_graph_slam_nodelet-6] process has finished cleanly
log file: /home/tong/.ros/log/985a32fa-d5e3-11e8-a625-80ce62f28ef3/hdl_graph_slam_nodelet-6*.log
^C[map2odom_publisher-7] killing on exit
[lidar2base_publisher-1] killing on exit
[velodyne_laser1-2] killing on exit
^

koide3, 2018-10-22:
Could you provide detailed information?
- OS and ROS version
- g2o version
- with a ros bag?

improve100, 2018-10-23:
@koide3 i solved it problem yesterday,by use this command:
sudo apt-get install ros-kinetic-libg2o
sudo cp -r /opt/ros/kinetic/lib/libg2o_* /usr/local/lib
sudo cp -r /opt/ros/kinetic/include/g2o /usr/local/include
rm build -rf
rm devel -rf
catkin_make -DCMAKE_BUILD_TYPE=Release

thank you for your awesome project.now it works well.