koide3/hdl_graph_slam #37: Cannot use own bag file

jer1234, 2018-10-25:
I am using VLP16 , and the .pcap I converted using velodyne_driver ,and then record as it .bag
But when I try to use it with  the your system nothing appear in Rviz , Seem like it did not detect my bag file. 
The mapping run smoothly with your  .bag file but not mine

koide3, 2018-10-25:
Could you check if the prefiltering nodelet is connected with the points topic using rqt_graph?

jer1234, 2018-10-28:
[image]

look like it is not connected, what should i do?

koide3, 2018-10-30:
Check the points topic name in your bag and remap "/velodyne_points" of prefiltering_nodelet following "Use hdl_graph_slam in your system" in README.