koide3/hdl_graph_slam #38: why some locations are sparse in new pointcloud map

improve100, 2018-11-15:
@koide3 
I have a question to ask, the following picture is the effect of my test with hdl_graph_slam. Above is the g2o graph, below is the laser point cloud map, the red circle indicates the corresponding position of the two maps, I don't understand what caused the position g2o map is very sparse. The point cloud map is also very sparse. Can you give me some advice?
[image]

koide3, 2018-11-15:
@improve100 ,
If you did it with a ROS bag, could you try to slow down the playback speed? I guess the CPU power was not enough, and the SLAM module dropped some frames...

koide3, 2018-11-19:
@improve100 ,
OK, your PC is definitely powerful enough for this package. Could you upload your launch file so that I can check some parameters?

improve100, 2018-11-20:
@koide3 
I also found another problem. In the scan matching node, ndt can easily calculate that the car is backward, that is, it falls into a local optimal solution. The actual car is moving forward. Especially in the corners.

<?xml version="1.0"?>
<launch>
  
  <arg name="nodelet_manager" default="hdl_nodelet_manager" />
  <arg name="enable_floor_detection" default="true" />
  <arg name="enable_gps" default="true" />
  <arg name="enable_imu_acc" default="false" />
  <arg name="enable_imu_ori" default="false" />
  <arg name="points_topic" default="/rslidar_points" />
  

  
  <node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 2 0 0 0 base_link rslidar 10" />
  <node pkg="tf" type="static_transform_publisher" name="imu2base_publisher" args="0 0 0 0 0 0 base_link imu 10" />

  
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>

  
  <node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="load hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">
    <remap from="/velodyne_points" to="$(arg points_topic)" />
    
    <param name="base_link_frame" value="base_link" />
    
    <param name="use_distance_filter" value="true" />
    <param name="distance_near_thresh" value="0.1" />
    <param name="distance_far_thresh" value="100.0" />
    
    <param name="downsample_method" value="VOXELGRID" />
    <param name="downsample_resolution" value="0.1" />
    
    <param name="outlier_removal_method" value="RADIUS" />
    <param name="statistical_mean_k" value="30" />
    <param name="statistical_stddev" value="1.2" />
    <param name="radius_radius" value="0.5" />
    <param name="radius_min_neighbors" value="2" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="load hdl_graph_slam/ScanMatchingOdometryNodelet $(arg nodelet_manager)" output="screen">
      <param name="points_topic" value="$(arg points_topic)" />
      <param name="odom_frame_id" value="odom" />
      <param name="keyframe_delta_trans" value="1.0" />
      <param name="keyframe_delta_angle" value="1.0" />
      <param name="keyframe_delta_time" value="1.0" />
      <param name="transform_thresholding" value="false" />
      <param name="max_acceptable_trans" value="1.0" />
      <param name="max_acceptable_angle" value="1.0" />
      <param name="downsample_method" value="NONE" />
      <param name="downsample_resolution" value="0.1" />
      
      <param name="registration_method" value="NDT_OMP" />
      <param name="ndt_resolution" value="1.0" />
      <param name="ndt_num_threads" value="0" />
      
      <param name="ndt_nn_search_method" value="DIRECT7" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="floor_detection_nodelet" args="load hdl_graph_slam/FloorDetectionNodelet $(arg nodelet_manager)" if="$(arg enable_floor_detection)">
    <param name="points_topic" value="$(arg points_topic)" />
    <param name="tilt_deg" value="0.0" />
    <param name="sensor_height" value="2.0" />
    <param name="height_clip_range" value="1.0" />
    <param name="floor_pts_thresh" value="512" />
    <param name="use_normal_filtering" value="true" />
    <param name="normal_filter_thresh" value="20.0" />
[... 45 lines cut ...]
    
    <param name="floor_edge_robust_kernel" value="NONE" />
    <param name="floor_edge_stddev" value="10.0" />
    
    
    <param name="odometry_edge_robust_kernel" value="Cauchy" />
    <param name="odometry_edge_robust_kernel_size" value="1.0" />
    <param name="loop_closure_edge_robust_kernel" value="Huber" />
    <param name="loop_closure_edge_robust_kernel_size" value="1.0" />
    <param name="use_const_inf_matrix" value="false" />
    <param name="const_stddev_x" value="0.5" />
    <param name="const_stddev_q" value="0.1" />
    <param name="var_gain_a" value="20.0" />
    <param name="min_stddev_x" value="0.5" />
    <param name="max_stddev_x" value="5.0" />
    <param name="min_stddev_q" value="0.1" />
    <param name="max_stddev_q" value="0.5" />
    
    <param name="graph_update_interval" value="3.0" />
    <param name="map_cloud_update_interval" value="10.0" />
    <param name="map_cloud_resolution" value="0.05" />
  </node>

  <node pkg="hdl_graph_slam" type="map2odom_publisher.py" name="map2odom_publisher" />
</launch>

improve100, 2018-11-20:
this is my launch file
[test.txt](https://github.com/koide3/hdl_graph_slam/files/2597656/test.txt)

koide3, 2018-11-21:
@improve100 ,
About the scan matching problem, maybe ndt_resolution is too small for outdoor. Larger resolutions (2m ~ 5m) are better for such environments. Please try to increase it.

I'm not sure what is the cause of the sparse map problem. But, could you try to change the parameters as follows:
- g2o_solver_type: gn_var_cholmod
- g2o_solver_num_iterations: 64
- graph_update_interval: 1.0