koide3/hdl_graph_slam #45: how to use pointcloud to further optimize generated map?

sume-cn, 2019-01-08:
On my dataset which was gathered in suburb environment about 5km long, multiple blur edges of same object was observed in generated pointcloud map. It seems the GPS constraint will cause the trajectory to bend. Since the optimization step and the point match step is totally seperated, this phenomenon can be expected.
In VSLAM, the reprojection error and GPS constraints may be optimized together, the result may be more reliable. 
So I think is there any way to optimize pointcloud matching and GPS together? or to use pointcloud further optimize the final result to get a more clear map?

koide3, 2019-01-08:
This system constructs and optimizes a pose graph which takes odometry (scan matching) and GPS constraints into account. So, actually, the optimization step and the matching step are not separated.
However, in case you give a large weight to GPS constraints, it breaks the local consistency (scan matching) due to the error on the GPS observations.
I thinks it's worth to try to make the GPS constraint weight smaller (increase gps_edge_stddev) and the scan matching weight larger (decrease (min|max)\_stddev_(x|q)). It will let the system put emphasis on the scan matching odometry constraints, and the point cloud quality may be improved.

sume-cn, 2019-01-09:
Thanks for your clear explanation, I'll try it immediately!

sume-cn, 2019-01-09:
It works! After increasing gps_edge_stddev, the final pointcloud map is much more sharp now.
BTW, why the gps_edge_stddev_xy is bigger than gps_edge_stddev_z ? To my knowledge, altitude is generally less accurate than longitude and latitude.
And, in the scan matching module, what does  x and q mean?
Thanks again.

closed 2019-08-08 by sume-cn