koide3/hdl_graph_slam #47: If I run in my computer and use VLP-16, warning: keyframes empty

MaxChanger, 2019-01-19:
I'm sorry to ask this seemingly simple question. But I haven't found a solution for a long time. 
I can run your bag file properly and save the map. However, when I wanted to build a map with my own velodyne, there were some problems. 

In your README.md, you mentioned that we should `<remap ……`, and I think if I use velodyne, I needn't modify it. So I connected the velodyne, and `roslaunch hdl_graph_slam hdl_graph_slam.launch`. But I can't see anything except `warning: keyframes empty`(In the lower left corner of the picture).  And if I want to see the topic use `rostopic echo /velodyne_points` , it's empty too. So I don't think the velodyne works. Do I need to set up something else?

[image]

I find a solution to this problem, I remove the line 15 in hdl_graph_slam.launch `<node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>`, and then start the velodyne by`roslaunch velodyne_pointcloud VLP16_points.launch calibration:=/home/sun/Desktop/VLP-16.yaml`, the point cloud data can be displayed, but after a while, new problems will arise.like this:

[image]

So I don't know how to run your code correctly with my velodyne. Can you provide some solutions? Thanks.

>Ubuntu 16.04 + Kinetic 
PCL 1.7 (In ROS)
g2o https://github.com/felixendres/g2o
I don't think it has much to do with these versions, because I can run bag properly.

koide3, 2019-01-21:
Hi @MaxChanger ,

> I find a solution to this problem, I remove the line 15 in hdl_graph_slam.launch

Yes, in case with a real velodyne instead of the rosbag, you need to remove the line to prevent that multiple nodelet managers are started. In addition, you have to set use_sim_time to false.

> the point cloud data can be displayed, but after a while, new problems will arise.like this:

It seems the frame name is not correctly provided. Could you check if the input cloud frame is correct by:
```bash
rostopic echo /velodyne_points/header
```

If the driver is working correctly, we should see something like:

```bash
---
seq: 1735
stamp: 
  secs: 1509348874
  nsecs: 626796000
frame_id: "velodyne"
---
seq: 1736
stamp: 
  secs: 1509348874
  nsecs: 726864000
frame_id: "velodyne"
```

Could you also let me see the tf tree using ```rosrun rqt_tf_tree rqt_tr_tree```?

MaxChanger, 2019-01-22:
The output of `rostopic echo /velodyne_points/header` :

[image]

I'm sorry, for some reasons, the velodyne lidar is not here now. I can only use bags I recorded before.
I use `roslaunch hdl_graph_slam hdl_graph_slam_501.launch` and my bag. the tf  like this:

[image]

But there are some warning( yellow ) in the terminal of rviz.

koide3, 2019-01-22:
I seems it is a bug of the marker generation code. In create_marker_array method, I allocated five markers, but only four of them are used and one is left empty. The empty marker does not affect anything but causes the yellow warnings. I'll fix this bug soon. For now, please ignore the warnings. Thank you for reporting :+1: 

https://github.com/koide3/hdl_graph_slam/blob/e26a6bfad620a7d93238f1d094fb69174b534e29/apps/hdl_graph_slam_nodelet.cpp#L610

GLKode, 2019-10-01:
Hi, on the same note, when I use the real velodyne instead of the bag file, the launch files crash immediately after starting velodyne's launch file. (it works perfectly with the bag file)
I have commented the mentioned line in the **hdl_graph_slam.launch** and ran it in a terminal, then in a new terminal i run roslaunch velodyne_pointcloud VLP-32C_points.launch, then immediately both launches are crashing.
providing the following error:

**hdl_graph_slam terminal**
[FATAL] [1569910260.413933249]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1569910260.433904877]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1569910260.443032293]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'

**Velodyne terminal**
[ERROR] [1569910260.413693306]: Failed to load nodelet [/scan_matching_odometry_nodelet] of type [hdl_graph_slam/ScanMatchingOdometryNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class hdl_graph_slam/ScanMatchingOdometryNodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_ros/data_odom_sync rtabmap_ros/data_throttle rtabmap_ros/disparity_to_depth rtabmap_ros/icp_odometry rtabmap_ros/imu_to_tf rtabmap_ros/obstacles_detection rtabmap_ros/obstacles_detection_old rtabmap_ros/point_cloud_aggregator rtabmap_ros/point_cloud_assembler rtabmap_ros/point_cloud_xyz rtabmap_ros/point_cloud_xyzrgb rtabmap_ros/pointcloud_to_depthimage rtabmap_ros/rgbd_odometry rtabmap_ros/rgbd_relay rtabmap_ros/rgbd_sync rtabmap_ros/rgbdicp_odometry rtabmap_ros/rtabmap rtabmap_ros/stereo_odometry rtabmap_ros/stereo_sync rtabmap_ros/stereo_throttle rtabmap_ros/undistort_depth stereo_image_proc/disparity stereo_image_proc/point_cloud2 velodyne_driver/DriverNodelet velodyne_laserscan/LaserScanNodelet velodyne_pointcloud/CloudNodelet velodyne_pointcloud/RingColorsNodelet velodyne_pointcloud/TransformNodelet

any idea what went wrong?

fujimo-t, 2019-10-03:
@GLKode 
The error in Velodyne terminal says that `hdl_graph_slam/ScanMatchingOdometryNodelet` nodelet is not found.

> According to the loaded plugin descriptions the class hdl_graph_slam/ScanMatchingOdometryNodelet with base class type nodelet::Nodelet does not exist.

But usually list of nodelets is not changed with or without of bag file.

I think you should check list of all nodelets by `rosrun nodelet declared_nodelets` command in terminal.

GLKode, 2019-10-03:
Thank you @fujimo-t for your reply, I have changed the atgs **load** to **standalone** for all nodelet in the launch file.  `node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="**load**` and it seems to be working now.

before this change the launch file was working when I was playing the bag file and when I ran the launch file of hdl_graph_slam with the launch file of velodyne it would produce the mentioned error. 
somehow running the velodyne launch is crashing the hdl_graph_slam launch when arg **load** is used.

fujimo-t, 2019-10-03:
I maybe notice cause of problem.
You should run `VLP-32C_points.launch` before `hdl_graph_slam.launch`.
>I have commented the mentioned line in the hdl_graph_slam.launch and ran it in a terminal, then in a new terminal i run roslaunch velodyne_pointcloud VLP-32C_points.launch, then immediately both launches are crashing.

**load** was failed because there is not nodelet manager when you launched `hdl_graph_slam.launch`.
The nodelet manager was launched in `VLP-32C_points.launch`.

GLKode, 2019-10-04:
Hi @fujimo-t I have tried that as well, i started the `VLP-32C_points.launch` before `hdl_graph_slam.launch` . by running the `VLP-32C_points.launch` everything works fine, once i start the `hdl_graph_slam.launch` both langues are crashing. if i dont launch the `VLP-32C_points.launch` and use the bag file instead of course i will uncomment the following line in `hdl_graph_slam.launch`
` <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>` everything works correctly.

To solve the problem as i mentioned in my earlier comment i have changed the atgs load to standalone for all nodelet in the launch file. node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="**load** and it seems to be working now. but i dont know how correct this solution is and wondering if I am the only one encountering this error? and what would be the correct way to solve it.

fujimo-t, 2019-10-04:
@GLKode I see.
Do you use different two catkin workspaces for velodyne_driver and hdl_graph_slam?

GLKode, 2019-10-04:
Yes, do i need to install the velodyne_driver inside hdl_graph_slam catkin workspace?

koide3, 2019-10-04:
Hi @GLKode,
I think the cause is the confliction of the nodelet manager names. hdl_graph_slam uses the same nodelet manager name as the one used in velodyne_driver so that they share point cloud data without costly communication. In case you run the SLAM with velodyne_driver, comment out the following line so the duplicated nodelet manager will not be started.
(Maybe I should mention this in README)

https://github.com/koide3/hdl_graph_slam/blob/9c360c10c776209e11ffb21f836405e6ccd0a3cd/launch/hdl_graph_slam_400.launch#L14

@fujimo-t
Thanks for your support :)

fujimo-t, 2019-10-04:
Hi @koide3 ,
I doubt another cause because he already commented out it.
>I have commented the mentioned line in the hdl_graph_slam.launch and ran it in a terminal,

@GLKode The nodelet manager maybe can't find path of hdl_graph_slam package if you don't `source <hdl_graph_slam's workspace/devel/setup.bash` in **velodyne terminal**.

GLKode, 2019-10-04:
Thanks @fujimo-t and @koide3 , Yes as @fujimo-t just mentioned I have already commented out that line, I believe that the problem must be the source, hence eveytime i want to run the hdl_graph_slam i need to `source ~/hdl_graph_slam/devel/setup.bash` then onlly it can find it, I have already saved the `source hdl_graph_slam's workspace>/devel/setup.bash` in the .bashrc file no idea why it cant find the source.
To be sure I have just installed the velodyne driver in the hdl_graph_slam workspace and it solved the problem.

fujimo-t, 2019-10-04:
@GLKode Congratulation!

The source problem seems to be related to workspace overlay but I'm not sure.
http://wiki.ros.org/catkin/Tutorials/workspace_overlaying#Chaining_catkin_workspaces

GLKode, 2019-10-04:
Thank you very much @fujimo-t  it was a big help and thanks for the workspace overlay hints, I do appreciate it .
May I use this opportunely and ask two more questions:
1- when changing the registration method in the launch file to **ICP** or any other registration method  do we need to make other changes as well? when I use ICP `<param name="registration_method" value="**ICP**" />` the system doesn't work : in terminal it shows the following and nothing happens afterwards:

```
[ INFO] [1570171498.628644047]: Initializing nodelet with 32 worker threads.
process[map2odom_publisher-8]: started with pid [19327]
downsample: NONE
registration: ICP
downsample: VOXELGRID 0.1
outlier_removal: RADIUS 0.5 - 2
construct solver... 
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
registration: ICP
```
2- The second question is why cant i see the **floor_plant** points in the rviz, when i uncheck all other point clouds and run the bag 400&501 files provided by hdl_graph_slam the floor_plan points are not shown.

fujimo-t, 2019-10-04:
These questions aren't related to original issue.
So you should find same issue from existed ones and if not found, open new issue.