koide3/hdl_graph_slam #51: saved pcd doesn't have intensities.

donrv, 2019-03-08:
Hi,
Is there any way to save Intensity values while saving pcd files?

koide3, 2019-03-15:
Hi @donrv ,
I guess intensity values are dropped when the point cloud is downsampled with pcl::Octtree.
https://github.com/koide3/hdl_graph_slam/blob/b97e2c1ab4877f2e9c89033b3f3f4d284940dd62/src/hdl_graph_slam/map_cloud_generator.cpp#L42

Maybe by changing this downsampling method, you can have a point clouds with intensities.

HappySamuel, 2019-03-21:
Hi @donrv 

I just found out an alternative way to same the pointcloud with intensity. But there is a pre-requistie for doing so. The generated pointcloud especially for mapping use must originally contain intensity values.

Then, you can try the following:

[image]

Hope my experience can help you.

Samuel

closed 2019-06-24 by donrv