koide3/hdl_graph_slam #53: Where to find the hdl_graph_slam/SaveMap.h ??

HappySamuel, 2019-03-20:
Hi,

I am curious where can i find this file (SaveMap.h). I have been searching it in all the directories, but i couldn't find them. I need to customize it for my own usage.

Best.
Samuel

koide3, 2019-03-20:
Hi @HappySamuel ,

The header file will be generated by catkin automatically. Once you do catkin_make, you should be able to find it in /your/catkin_ws/devel/include/hdl_graph_slam

HappySamuel, 2019-03-21:
Hi @koide3 

Yeah, i found it. Thank you very much. By the way, i found an alternative way to save the PointCloud map. It's through pcl_ros package with command as figure below:

[image]

closed 2019-03-21 by HappySamuel