koide3/hdl_graph_slam #58: How to do benchmark on KITTI datasets

HongweiSunny, 2019-05-16:
Thx for your great working!
I want to do some benchmark based on KITTI datasets to check the accuracy of this method.
But I do not know how to get the trajectory of mapping.
Looking forward to your help.
Thx in advance.

koide3, 2019-05-27:
Hi @HongweiSunny ,
You may need to convert KITTI to rosbag. I'm not sure, but I think there should be some scripts to do that on github...

mhaboali, 2020-04-17:
I have now a rosbag that has all Kitti sequence 08 from the odometry dataset, but when I ran it with hdl_---_kit.launch with configuring the static_tf_publisher for Velodyne and cameras data, it didn't work as properly. 

Most properly there's an issue with the tf because of time difference.
[image]

And here's the TF tree
[image]

And from RVIZ I got those errors as shown below
[image]