koide3/hdl_graph_slam #65: [velodyne_nodelet_manager-3] process has died [pid 3601, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/mdg/.ros/log/0d1fa298-a189-11e9-b6fd-24fd524019c1/velodyne_nodelet_manager-3.log]. log file: /home/mdg/.ros/log/0d1fa298-a189-11e9-b6fd-24fd524019c1/velodyne_nodelet_manager-3*.log

gao0511, 2019-07-08:
HI r u still there? i still have a big problem that #4 can not solve..

gao0511, 2019-07-08:
[FATAL] [1562595065.585840784]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1562595065.585961188]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1562595065.586020264]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-3] process has died [pid 3947, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/velodyne_nodelet_manager-3.log].
log file: /home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/velodyne_nodelet_manager-3*.log
[prefiltering_nodelet-4] process has died [pid 3949, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager /velodyne_points:=/velodyne_points __name:=prefiltering_nodelet __log:=/home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/prefiltering_nodelet-4.log].
log file: /home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/prefiltering_nodelet-4*.log
[scan_matching_odometry_nodelet-5] process has died [pid 3954, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/scan_matching_odometry_nodelet-5.log].
log file: /home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/scan_matching_odometry_nodelet-5*.log
[hdl_graph_slam_nodelet-7] process has died [pid 3961, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/hdl_graph_slam_nodelet-7.log].
log file: /home/mdg/.ros/log/38bf7012-a18a-11e9-b6fd-24fd524019c1/hdl_graph_slam_nodelet-7*.log

gao0511, 2019-07-08:
#4 do not work in my system... ubuntu 18.04 ros melodic

gao0511, 2019-07-08:
thanks a lot for this

koide3, 2019-07-10:
Hi @gao0511 ,
What happens if you lanuch the nodelets in standalone mode? Which nodelet crashes?

huayoufeng, 2019-07-19:
I met the same problem, how to solve it?

huayoufeng, 2019-07-19:
I have solved this problem. Re-install g2o library.

zhouzhibo0117, 2019-08-01:
Are you using pcl-1.8? Try to remove pcl-1.8 in your system and use pcl-1.7 (ROS default) instead, then the problem will be solved.

lfriendsl, 2019-09-09:
@zhouzhibo0117 @koide3 I have the same problem,  somehow can't install pcl-1.7 on Ubuntu 18 it outputs errors, using the 
**sudo apt-get install ros-melodic-geodesy ros-melodic-pcl-ros ros-melodic-nmea-msgs ros-melodic-libg2o** pcl1.8 is automatically installing.

@huayoufeng could you please let me know how exactly you did it? I am trying to reinstall g2o, but the error exist. however I must add LIST( APPEND CMAKE_MODULE_PATH ~/g2o/cmake_modules ) to my CMakeList.txt, otherwise it cant find the g2o package.
Thanks in advance..

wayneasimo, 2020-10-22:
> #4 do not work in my system... ubuntu 18.04 ros melodic

hi, have you solved it? I seem to meet the same problem with you.

wayneasimo, 2020-10-22:
> I have solved this problem. Re-install previous g2o library.

hi, I met the same problem. which version of g2o do you use?

Dysonsun, 2021-05-06:
add these to your cmakelist.txt
```
install(TARGETS
  prefiltering_nodelet
  ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

AleksandrKashirin, 2022-07-26:
Which one "CMakiLists.txt" do you mean?