koide3/hdl_graph_slam #76: Changing registeration method

GLKode, 2019-10-04:
Thank you very much @fujimo-t  it was a big help and thanks for the workspace overlay hints, I do appreciate it .
May I use this opportunely and ask one more questions:
-Wwhen changing the registration method in the launch file to **ICP** or any other registration method  do we need to make other changes as well? when I use ICP `<param name="registration_method" value="**ICP**" />` the system doesn't work : in terminal it shows the following and nothing happens afterwards:

```
[ INFO] [1570171498.628644047]: Initializing nodelet with 32 worker threads.
process[map2odom_publisher-8]: started with pid [19327]
downsample: NONE
registration: ICP
downsample: VOXELGRID 0.1
outlier_removal: RADIUS 0.5 - 2
construct solver... 
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
registration: ICP
```

_Originally posted by @GLKode in https://github.com/koide3/hdl_graph_slam/issues/47#issuecomment-538270766_

koide3, 2019-10-07:
Hi @GLKode ,
No, you need to change only "registration_method" to use another registration method. Do other methods such as GICP work?

GLKode, 2019-10-07:
Hi @koide3, 
Thank you for your reply. I have tried all the mentioned methods in the launch file, the GICP as well, I have changed the registration method in two places in the launch file. The **NDT_OMP** is working correctly, the **GICP_OMP** works for few seconds and other don't work at all. Do we need to change the registration parameters?

this is the result showing in terminal when using the **GICP** method:
```
process[lidar2base_publisher-2]: started with pid [29725]
process[velodyne_nodelet_manager-3]: started with pid [29726]
process[prefiltering_nodelet-4]: started with pid [29732]
process[scan_matching_odometry_nodelet-5]: started with pid [29733]
process[hdl_graph_slam_nodelet-6]: started with pid [29735]
[ INFO] [1570436272.660019325]: Initializing nodelet with 32 worker threads.
process[map2odom_publisher-7]: started with pid [29784]
downsample: NONE
registration: GICP
construct solver... 
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
registration: GICP
downsample: VOXELGRID 0.1
outlier_removal: RADIUS 0.5 - 2

```

koide3, 2019-10-08:
What is the version of PCL in your environment? I found that the older GICP implementation in PCL has a problem on handling initial guess which has been fixed in the recent PCL (1.8 or later). In my environment (18.04 & melodic), GICP works well while GICP_OMP does not because it was derived from PCL 1.7. (I would update GICP_OMP later)

> why same have not been set for ICP or GICP ?
While the accuracy and robustness of NDT deeply depends on the hyper parameters, ICP variants are relatively less-sensitive to the parameters. I saw that GICP works well with the default parameters, and thus I didn't change them. But, there would be room to improve it by tweaking the parameters, and if you want I can make them configurable from the launch file.

GLKode, 2019-10-08:
Thank you @koide3 for your reply.
My environment is also Ubuntu (18.04 & melodic) and I have tried to install PCL 1.7 but my attempts were not successful, I was encountering errors installing it from sources and by running this command `sudo apt-get install ros-melodic-geodesy ros-melodic-pcl-ros ros-melodic-nmea-msgs ros-melodic-libg2o` it would install the 1.8 instead of 1.7, **so now I have 1.8 version** and therefore I have changed the `find_package(**PCL 1.7** REQUIRED`) in CMakeLists.txt to `find_package(**PCL 1.8** REQUIRED)`
 while compiling the hdl_graph_slam i encounter the following warning message as well.
```
usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here

```
Which version of PCL do you recommend? and if i should install the 1.7 then would you please let me know how to install it on Ubuntu 18.04?

> and if you want I can make them configurable from the launch file.
Thank would be great if you could do that.

koide3, 2019-10-08:
It seems the message is just a warning not an error. You can just ignore it.

> Which version of PCL do you recommend?

I recommend PCL 1.8 which you can install with apt-get: on Ubuntu  18.04:
```bash
sudo apt install ros-melodic-pcl-ros
```

> it would install the 1.8 instead of 1.7, so now I have 1.8 version and therefore I have changed the find_package(PCL 1.7 REQUIRED) in CMakeLists.txt to find_package(PCL 1.8 REQUIRED)

You don't need to change it since it's **minimum** required version.

GLKode, 2019-10-08:
Hi @koide3 . I believe I have the version 1.8 already installed, but I have run the command you just mentioned and I get the following message:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-melodic-pcl-ros is already the newest version (1.6.2-0bionic.20190808.211311).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
in that case what do you think is the reason that the registration methods ICP and GICP not to work? in my environment? as you have mentioned with version 1.8 the suppose to work.

koide3, 2019-10-08:
Hi @GLKode ,
What happens if you playback the rosbag slowly like:
```bash
rosbag play -r 0.1 --clock rosbag.bag
```
The original ICP and GICP are slow and the scan matching may drop some frames. Could you also try to subscribe to /odom topic to check if the odometry estimation is working properly?

GLKode, 2019-10-09:
Hi @koide3 ,
Thank you for the reply, when the bag file is played slowly it works, seems that as you pointed out the original ICP and GICP are very slow, the question here is, how to use these registration methods when subscribing to the **velodyne driver directly instead of running the bag files**, it would not be possible to use the ICP or GICP in such case, or is there a way to run the velodyne points  also slower?

koide3, 2019-10-10:
Hi @GLKode ,
Slower scan matching makes the SLAM quality drastically worse. I thinks the only way is to use GICP in PCL1.8 or NDT_OMP  for now. In my environment they run faster than real-time.

GLKode, 2019-10-10:
Hi @koide3,
Thank you very much for the information,

> But, there would be room to improve it by tweaking the parameters, and if you want I can make them configurable from the launch file.

I would appreciate it if you could provide those configurable parameters, or else let me know which parameters are to be tweaked in ICP, or GICP. 

Thank you.

koide3, 2019-10-10:
Hi @GLKode ,
I made scan matching parameters configurable as rosprams. Can you pull **devel** branch to check them out? You can find configurable params in hdl_graph_slam.launch.

GLKode, 2019-10-10:
Thank you very much @koide3 , i see that you have done the pull request as well.

closed 2019-10-10 by GLKode