koide3/hdl_graph_slam #80: start pose far away from map frame ???

chennuo0125, 2019-10-11:
[image]
  
https://github.com/koide3/hdl_graph_slam/blob/e5f8b271b29b1b75e242f621d8ea22e019008802/apps/hdl_graph_slam_nodelet.cpp#L203  
above code has fixed first pose, why start pose far away from map frame ???

koide3, 2019-10-12:
Hi @chennuo0125 ,
The code just above that line decides if the first node is fixed.
https://github.com/koide3/hdl_graph_slam/blob/e5f8b271b29b1b75e242f621d8ea22e019008802/apps/hdl_graph_slam_nodelet.cpp#L201

Make "fix_first_node" in the launch file true to enable fixing the fist node.

chennuo0125, 2019-10-12:
oh sorry, i didnt see it clearly, thank you very much :) :) :)

closed 2019-10-12 by chennuo0125