koide3/hdl_graph_slam #82: nodelet crash

sintax2con, 2019-10-17:
> munmap_chunk(): invalid pointer
[velodyne_nodelet_manager-4] process has died [pid 5836, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/bzl-kangj/.ros/log/5e14da2c-f0b9-11e9-b8d4-b068e687c4ab/velodyne_nodelet_manager-4.log].
log file: /home/bzl-kangj/.ros/log/5e14da2c-f0b9-11e9-b8d4-b068e687c4ab/velodyne_nodelet_manager-4*.log

this problem suddenly occured, it worked fine before. I tried to reinstall g2o as   #4 ,but it doesn't help.
i use melodic vesion .
thanks a lot!

sintax2con, 2019-10-17:
> _opt_ros_melodic_lib_nodelet_nodelet.1000.crash
and i found a crash in /var/crash

koide3, 2019-10-18:
Hi @sintax2con ,
Could you try to launch the nodelets in the standalone mode to see which nodelet causes the crash:

```
# modify the following line:
<node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="load hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">

# to:
<node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="standalone hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">

# for each nodelet
```

sintax2con, 2019-10-18:
> Hi @sintax2con ,
> Could you try to launch the nodelets in the standalone mode to see which nodelet causes the crash:
>
> ```
> # modify the following line:
> <node pkg="nodelet" type="nodelet" [...]

Hi @koide3 ,
i change all the args to standalone 
i think i maybe something wrong with the nodelet_manager
once i roslaunch (without play bag ),this node  crashed

> <pre><b>process[lidar2base_publisher-2]: started with pid [14104]</b>
<b>process[imu2base_publisher-3]: started with pid [14105]</b>
<b>process[velodyne_nodelet_manager-4]: started with pid [14107]</b>
<b>process[prefiltering_nodelet-5]: started with pid [14116]</b>
Your usage: 
/opt/ros/melodic/lib/nodelet/nodelet standalone __name:=velodyne_nodelet_manager __log:=/home/sintax/.ros/log/16ce6ad8-f164-11e9-bb1b-d8cb8a969461/velodyne_nodelet_manager-4.log 
nodelet usage:
nodelet load pkg/Type manager [--no-bond]  - Launch a nodelet of type pkg/Type on manager manager
nodelet standalone pkg/Type   - Launch a nodelet of type pkg/Type in a standalone node
nodelet unload name manager   - Unload a nodelet by name from manager
nodelet manager               - Launch a nodelet manager node
<b>process[scan_matching_odometry_nodelet-6]: started with pid [14118]</b>
type is hdl_graph_slam/PrefilteringNodelet
<font color="#CC0000">[velodyne_nodelet_manager-4] process has died [pid 14107, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone __name:=velodyne_nodelet_manager __log:=/home/sintax/.ros/log/16ce6ad8-f164-11e9-bb1b-d8cb8a969461/velodyne_nodelet_manager-4.log].</font>
<font color="#CC0000">log file: /home/sintax/.ros/log/16ce6ad8-f164-11e9-bb1b-d8cb8a969461/velodyne_nodelet_manager-4*.log</font>
<b>process[floor_detection_nodelet-7]: started with pid [14120]</b>
type is hdl_graph_slam/ScanMatchingOdometryNodelet
<b>process[hdl_graph_slam_nodelet-8]: started with pid [14129]</b>
type is hdl_graph_slam/FloorDetectionNodelet
<b>process[map2odom_publisher-9]: started with pid [14131]</b>
</pre>

sintax2con, 2019-10-19:
solve it , i add a passthrough filter in the prefilterring nodelet, it worked fine before, but now it doesn't

closed 2019-10-19 by sintax2con