koide3/hdl_graph_slam #84: [pcl::NormalEstimation::compute] input_ is empty!

facontidavide, 2019-10-18:
Hi,

I am trying outdoor mapping with my rosbag (rslidar) and  hdl_graph_slam_400.launch

I of course remapped the rslidar_point message, but I get the following message:

                  [pcl::NormalEstimation::compute] input_ is empty!

Any clue? Do you want me to share the rosbag to reproduce the error?

sintax2con, 2019-10-19:
unable to detect the floor， maybe too few points after prefilterring 
check the prefilterring nodelet

facontidavide, 2019-10-19:
I am using a 16 rays Robosense laser and the field of view is limited to +/- 110 degrees.

I guess I can edit the parameter **floor_pts_thresh** accordingly.

If I can make a suggestion, it would be nice if the error message is a little more informative :)

closed 2019-10-19 by facontidavide

facontidavide, 2019-10-19:
For the record, it worked. Mapping is ok-ish, whilst localization is a bit off, but this is another problem.

ipinyulun, 2020-09-16:
hi @facontidavide 
what did u edit to have it works?
i try to decrese and increase floor_pts_thresh but its not working

arjunskumar, 2022-02-23:
@ipinyulun are you able to solve this?

Yu1107, 2022-03-09:
* in hdl_graph_slam/apps/floor_detection_nodelet.cpp
  ```
    // too few points for RANSAC
    if(filtered->size() < floor_pts_thresh) {
      std::cout << "filtered: " << filtered->size() << std::endl;
      return boost::none;
    }
  ```
filtered->size() always 0 so input_ is empty!

and changing "enable_floor_detection" in your launch file to "false"

arjunskumar, 2022-03-09:
@Yu1107 thanks, I was able to work in our small office environment.

ElFloco, 2023-03-06:
@Yu1107 you are right, the array is always empty, because if you have a sensor height greater than `height_clip_range`, then the floor is not considered

I edited following line in `floor_detection_nodelet.cpp:`
`filtered = plane_clip(filtered, Eigen::Vector4f(0.0f, 0.0f, 1.0f, sensor_height - height_clip_range), true);`

to:
`filtered = plane_clip(filtered, Eigen::Vector4f(0.0f, 0.0f, 1.0f, -height_clip_range), true);`

So points below are considered aswell. Now the floor detection works.