koide3/hdl_graph_slam #86: Outlier removal    NONE, RADIUS, or STATISTICAL

GLKode, 2019-10-21:
Hi @koide3 ,
I have a question regarding outlier removal, in prefiltering_nodelet, you have implemented two methods for the outlier removal but in all launch files only used the RADIUS method. I wonder in which cases shall we use STATISTOCAL method, and would it not bet better to use both methods simultaneously, to filter more noise?

koide3, 2019-10-23:
Hi @GLKode ,
Since the parameters of radius outlier removal (radius and # of neighbors) are more intuitive than the ones of statistical method, I prefer to use it. Although I expect they don't show a big difference on 3D LIDAR data, statistical one could be more reasonable in some cases. The following thread would be helpful to decide which one is more suitable for your application:
http://www.pcl-users.org/Methods-for-outlier-removal-td4036551.html

GLKode, 2019-10-23:
Thank you @koide3 for the clarification.

closed 2019-10-23 by GLKode