koide3/hdl_graph_slam #87: Finer mapping

arixrobotics, 2019-10-23:
Hello, 
Thank you for the fantastic package! I've tried with a VLP16 and it worked in real-time nicely. 

However, I am exploring the different parameters available. In particular, can I make the resulting mapping 'finer' / more dense? I'm mapping a small area, and would want the map to appear dense / refined.

koide3, 2019-10-23:
Hi @arixrobotics ,
You may want to change the following params:
- "prefiltering_nodelet/downsample_resolution" (In case resolution < 0.1, use APPROX_VOXELGRID instead of VOXELGRID)
- "hdl_graph_slam_nodelet/keyframe_delta_trans" (By making the keyframe interval small, you can yield a dense point cloud, but too small value may affect the processing speed)

If you want an extremely dense map, it would be worth considering to interpolate keyframe poses with odometry information and accumulate all input point clouds.

arixrobotics, 2019-10-24:
Thanks @koide3 , that worked. 
Can I know where can I learn how to tune all the available parameters? I know that different environment (urban vs lots of trees vs indoor etc) requires different set of parameters for the mapping to be effective. 
Do you have like a list of what each parameter means or something like that? Or is it in the code? Where can I understand them? 
I can also help to compile this list of parameters if you want.

koide3, 2019-10-26:
All the configurable parameters are listed in the launch file, but I have to say it's not very well documented yet... I think the documentation should be enhanced more, for instance, by putting a list of params and descriptions in wiki). I would be grateful if you could help to compile such a list.