koide3/hdl_graph_slam #89: map drift a lot,loop not found

liuyf5231, 2019-10-29:
Hi koide3 ,

I tested hdl graph slam by robosense 16 lidar, but i found that there is something wrong with the map. 

```
--- loop detection ---
num_candidates: 8
matching........ done
best_score: 36.815    time: 0.000[sec]
loop not found...
```

tf between map and odom, 
[image]

```
At time 1572313024.112
- Translation: [-5.383, -2.814, 16.107]
- Rotation: in Quaternion [-0.001, -0.031, -0.034, 0.999]
            in RPY (radian) [-0.001, -0.062, -0.068]
            in RPY (degree) [-0.034, -3.564, -3.881]
```

liuyf5231, 2019-10-29:
[image]

liuyf5231, 2019-10-29:
[image]

koide3, 2019-10-29:
Hi @liuyf5231 ,
NDT-based methods may not work well with 16-line LIDARs because it is not very robust to sparse point clouds. Could you try to use GICP or GICP_OMP instead of NDT_OMP (by changing parameters in the launch file)? If you are using ROS melodic, GICP should work well (PCL on Kinetic or earlier has a bug). You can also try to slow down the rosbag playback speed:
```bash rosbag play --clock -r 0.2 bagfilename.bag```

liuyf5231, 2019-10-29:
> Hi @liuyf5231 ,
> NDT-based methods may not work well with 16-line LIDARs because it is not very robust to sparse point clouds. Could you try to use GICP or GICP_OMP instead of NDT_OMP (by changing [...]

ok, I will try.

koide3, 2019-10-30:
In that case, maybe you can use NDT_OMP since the target point cloud (map cloud) is dense.

closed 2019-11-09 by liuyf5231