koide3/hdl_graph_slam #90: how to transform the map to utm frame?

liuyf5231, 2019-11-13:
Hi koide3,
how to transform the map to UTM frame? or how to get the origin of the map frame in UTM?

koide3, 2019-11-14:
Hi @liuyf5231 ,
You can use "hdl_graph_slam/dump" service which saves all the internal map data. It yields a file "zero_utm" which describes the position of the origin of the map in the UTM coordinate. You can just add these values to each point in the map.

liuyf5231, 2019-11-14:
> Hi @liuyf5231 ,
> You can use "hdl_graph_slam/dump" service which saves all the internal map data. It yields a file "zero_utm" which describes the position of the origin of the map in the UTM coordinate. You can just add these values to each point in the map.

yes, I know zero_utm.
but send a static transform between map frame and utm frame,   yaw pitch roll or quaternion is also needed.

If I want to use the map generated by hdl graph slam, first I need to transform my init gps to map frame.

koide3, 2019-11-14:
The map is optimized so that XYZ axes match with easting-nothing-altitude axes. So, you don't need to consider orientation and just add zero_utm to the points.

liuyf5231, 2019-11-15:
ok, thank you.

closed 2019-11-15 by liuyf5231