koide3/hdl_graph_slam #94: MarkerArray msg is empty

ricber, 2019-11-25:
Hi, 
I'm having a problem on the following configuration:
Ubuntu 16.04
ROS Kinetic

The problem is that the MarkerArray message has one of the elements of the array with an empty frame_id. See the following terminal output where I've highlighted in bold the empty frame_id message:
`markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1574692887
        nsecs: 223545959
      frame_id: "map"
    ns: "nodes"
    id: 0
    type: 7
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    scale: 
      x: 0.5
      y: 0.5
      z: 0.5
    color: 
      r: 0.0
      g: 0.0
      b: 0.0
      a: 0.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: 
      - 
        x: 0.0
        y: 0.0
        z: 0.0
    colors: 
      - 
        r: 1.0
        g: 0.0
        b: 0.0
        a: 1.0
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1574692887
        nsecs: 223545959
      frame_id: "map"
    ns: "edges"
    id: 1
    type: 5
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    scale: 
      x: 0.05
      y: 0.0
      z: 0.0
    color: 
      r: 0.0
      g: 0.0
      b: 0.0
      a: 0.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      **frame_id: ''
    ns: ''**
    id: 0
    type: 0
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.0
      y: 0.0
      z: 0.0
    color: 
      r: 0.0
      g: 0.0
      b: 0.0
      a: 0.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1574692887
        nsecs: 223545959
      frame_id: "map"
    ns: "loop_close_radius"
    id: 0
    type: 2
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    scale: 
      x: 2.0
      y: 2.0
      z: 2.0
    color: 
      r: 1.0
      g: 0.0
      b: 0.0
      a: 0.300000011921
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1574692887
        nsecs: 223545959
      frame_id: "map"
    ns: "imu"
    id: 4
    type: 7
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    scale: 
      x: 0.75
      y: 0.75
      z: 0.75
    color: 
      r: 0.0
      g: 0.0
      b: 0.0
      a: 0.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
---
`
I use the hdl_graph_slam.launch file with the nodelet_manager line commented out as I use the velodyne_driver.
So, I first start hdl_graph_slam, then Rviz, the the velodyne_driver. For some tens of seconds I correctly see the map building on Rviz but after a while I get this error message from Rviz:
`[ WARN] [1574694951.774019641]: MessageFilter [target=map ]: Discarding message from [unknown] due to empty frame_id.  This message will only print once.
[ WARN] [1574694951.774175074]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
`
However, no errors are printed from hdl_graph_slam nor from the velodyne_nodelet_manager

Can you help me? Thank you :)

koide3, 2019-11-26:
Hi @ricber ,
Yes, it's a known issue caused by markers.markers[2] which is always empty. Since it was not very serious (just shows warnings) and I was a bit busy last year, I forgot to fix it. I'll update the code soon. Thank you for reporting it :)

jkoubs, 2022-08-05:
Hello were you able to resolve this error ? Thank you

Franzisdrak, 2023-01-23:
I am having the same issue. But i don't use the graph slam. Might be due to something in the source code of the MarkerArray?