koide3/hdl_graph_slam #95: Map is not saved correctly

ricber, 2019-12-03:
Hi, 
I'm testing hdl with my own data. When I do the mapping with `hdl_graph_slam` the map is correctly showed parallel to the ground.
[image]
 
Then, I call the `save_map` service to obtain the pcd file. However, the saved map is no more correctly referenced to the `map` frame but it has a wierd rotation and translation.
[image]

Do you know what's causing the problem? I'm just using the `velodyne_points` data, no GPS, no IMU or anything else (ground detection is enabled).

Thanks.

koide3, 2019-12-04:
Hi @ricber ,
Do you see the same problem if you open the PCD with some point cloud viewers (e.g., pcl_viewer or CloudCompare)?

ricber, 2019-12-04:
Yep, I see the same problem with the pcl_viewer

ricber, 2019-12-04:
Yep, I see the same problem with the pcl_viewer.

**EDIT:** I tried again but I cannot reproduce the error. I don't know what happened at that time. I close the issue. Thank you anyway :)

ricber, 2020-04-04:
Actually, the problem is solved by setting the `enable_floor_detection` to true. If I emove the ground plane constraint, the problem is still there. Do you have any idea?

ricber, 2020-04-04:
Closed because of #113

closed 2020-04-04 by ricber