koide3/hdl_graph_slam #98: wrong map orientation with a tilted sensor

GLKode, 2019-12-11:
Hi @koide3 ,
I have tested the system with a 10 and 20 degree tilted sensor, to detect the floor points and more details in the floor, 
`   <param name="tilt_deg" value="10.0" />
    <param name="sensor_height" value="1.0" />
    <param name="height_clip_range" value="0.3" />
    <param name="floor_pts_thresh" value="256" />
    <param name="use_normal_filtering" value="true" />
    <param name="normal_filter_thresh" value="30.0" />`
 when I tilt my sensor, and set the above parameter in the lunch file the  system is detecting the floor points, but all point are oriented in exact opposite directed and the whole map is then tilted as well, in the opposite direction, see screenshots below:
**floor points**
[image]
**mappoints**
[image]
,
any idea what went wrong? and how to fix it?

 I also would like to ask how do we calculate the **height_clip_range value**.
Thanks in advance

koide3, 2019-12-13:
Hi @GLKode ,
What is the TF transformation you defined between "velodyne" and "base_link" frames? If you put some rotation there, the tilted map phenomenon could be a bug. I'm not very confident if I properly implemented rotation compensation in the floor plane handling code. If you don't mind, could you share (a part of) your bag file so that I can do some debugging?

Regarding **height_clip_range**:
The floor plane detector first extracts points in the z-range ``` |p.z - sensor_height| < height_clip_range ```, then applies RANSAC to find the plane coefficients. So, tune height_clip_range (and sensor_height & tilt_deg) so that all the floor points fall in this range and are detected as the floor plane.

GLKode, 2019-12-13:
Hi @koide3 , thanks for your reply.
I have tested the system on 20 degree and changed the  TF transformation parameters as well, `<node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 0 0 **0.33** 0 base_link velodyne 10" />`, it starts the map normally the it starts to deviate and eventually accumulate the error very fast and when turning then the map is unrecognizable. 

In the following is a sample bag file(filtered points), with 20-degree sensor tilt, and 1-meter height.:
[https://gofile.io/?c=P454CG](url)

looking forward to see your feedback!
Thank you in advance!

VicSegers, 2020-03-13:
I have the same problem, but it is not a static tilt. I use this slam on a drone and when it flies it tilts forward and the map gets projected at that angle.

koide3, 2020-03-18:
Hi guys,
I try to make spare time to fix this bug in next week. Although I already have GLKode's bag file, if you have another (small) rosbag to reproduce this problem, can you share it with me?

VicSegers, 2020-03-18:
Hey koide3,
We are currently using another SLAM algorithm, because of other restrictions. But thank you for your feedback!

GLKode, 2020-05-29:
Hi @koide3 , any news on this front? am still having the same problem. Did you need a smaller rosbag file to reproduce the error?

AchmadFathoni, 2021-04-10:
> I have the same problem, but it is not a static tilt. I use this slam on a drone and when it flies it tilts forward and the map gets projected at that angle.

Was your lidar attached on the drone vertically or horizontally?

ManChrys, 2021-07-01:
Hello guys,

i have the same problem when i tried to use my rs16 lidar (almost 45 degrees).

[image]
[image]

tf is  <node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 0 0 0 -40 base_link rslidar 10" />

because i think that the last number is roll ( x axis). I tried also the other numbers but nothing happens.

Im using velodyne on 45 degrees for a mobile mapping and when i tried the same tf it worked properly.

himhan34, 2022-12-08:
i am doing the same thing for the velodyne lidar, 
and also i am trying with 45 degrees. so the thing that you have done is using tf? 

using, this? 
<node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 0 0 0 -40 base_link rslidar 10" /> 
thanks for reading