koide3/hdl_graph_slam #99: Run launch file corrupted immediately.

YangSiri, 2019-12-13:
HI, @koide3 

Once i run like this _roslaunch hdl_graph_slam hdl_graph_slam.launch_. Corruption like this:

[image]

Then i tried other launch files, it is much like the same 

[image]

Looking forward to your help.
THANKS

koide3, 2019-12-13:
Hi @YangSiri ,
If you are using self-built g2o library, try libg2o debian package instead: ```sudo apt install ros-melodic-libg2o```

This thread could help you: https://github.com/koide3/hdl_graph_slam/issues/4

YangSiri, 2019-12-13:
Thanks, @koide3 

My environment is ubuntu16.04 kinetic. Will the libg2o you mentioned conflict with my source-built g2o?

koide3, 2019-12-13:
It may conflict with your g2o. If you don't like it, you can give the docker image a try: https://github.com/koide3/hdl_graph_slam#hdl_graph_slam-in-docker