koide3/hdl_graph_slam #100: does hdl graph slam support GPS available signal?

ZOUJIASHUAI, 2019-12-16:
I have a RTK GPS module and it cannot work in indoor environment,but when the robot at indoor Environment it will send a gps unavailable signal to the ros,so is there anyway I can feed the gps unavailable signal to the hdl graph slam?

koide3, 2019-12-16:
Hi @ZOUJIASHUAI ,
This package does not care GPS availability and adds all fed GPS data to the pose graph. You need to write a small node which filters out invalid GPS data and feeds the rest to the SLAM node. Another way it to modify the SLAM code so that it receives your custom GPS messages and rejects invalid ones.
https://github.com/koide3/hdl_graph_slam/blob/03f93a6832478c635462f1a25f01b4d7f7c765a9/apps/hdl_graph_slam_nodelet.cpp#L256

ZOUJIASHUAI, 2019-12-16:
> Hi @ZOUJIASHUAI ,
> This package does not care GPS availability and adds all fed GPS data to the pose graph. You need to write a small node which filters out invalid GPS data and feeds the rest to the [...]

       sorry for the lack of  theoretical knowledge I'm don't understand how the slam code use the gps info,for example, can I just stop publish the gps info if the robot enter an indoor  environment(50m*50m),and keep send gps info when the robot go out ?
thanks

ZOUJIASHUAI, 2019-12-17:
Hi @koide3,
  for example, can I just stop publish the gps info if the robot enter an indoor  environment(50m*50m),and keep send gps info when the robot go out ?

koide3, 2019-12-17:
Yes, you can stop feeding GPS data anytime you want, then the SLAM constructs the map without GPS until you resume feeding GPS.