koide3/hdl_graph_slam #105: Shift in the Z direction

Johntec, 2020-02-11:
Hi @koide3, thanks for the good work, I have installed and tested your system it works perfectly when tested on your bag files, however when using the same launch files with my bag files there will be a shift in the Z-axis that accumulate itself in time and shifts the whole map. I believe this has been also noticed by others; any idea what causes this problem and how to solve it?

Thank you

koide3, 2020-02-14:
Hi @Johntec ,

This phenomenon can be suppressed by set keyframe_delta_time parameter to a large value. See https://github.com/koide3/hdl_graph_slam/issues/74

Johntec, 2020-02-20:
Thanks @koide3 for the reply, I have tried the keyframe_delta_time before creating the issue, it didn't help, this shift is more obvious when running the scan matching odometry nodelet, somehow its effect is reduced in graph slam nodelet, but still noticeable, for my case, I could minimiyźe its effect by changing the NDT resolution, but finding the correct value is also difficult since I am not aware of a correct formula for it, for different scans different values are required. 
This phenomenon is also increased when the laser scanner is stationary, is there a way to add ZUPT option to reduce this effect?