koide3/hdl_graph_slam #110: What is the meaning of the anchor_node and anchor_edge?

wwtinwhu, 2020-03-25:
Hi,  I can't figure out the different about using anchor_node and anchor_edge to fix first node and just to set first_node.setFixed(True) instead of using anchor.

Would really appreciate any help. Thanks!

koide3, 2020-03-26:
Hi @wwtinwhu ,
setFixed(True) hardly fixes the first node at the initial estimate (i.e., at the origin). However, it makes inconsistency when we use some constraints, for instance, acceleration vector and floor plane constraints may want to rotate or shift the first node. The anchor aims to loosely fix the first node around the origin.

The anchor implementation in hdl_graph_slam is not completed. Maybe it's better to improve it in the way used in interactive_slam:
https://github.com/SMRT-AIST/interactive_slam/blob/33ba5415dfcfeb3580593910c10ae05831bebe99/src/hdl_graph_slam/interactive_graph.cpp#L376

wwtinwhu, 2020-03-26:
> Hi @wwtinwhu ,
> setFixed(True) hardly fixes the first node at the initial estimate (i.e., at the origin). However, it makes inconsistency when we use some constraints, for instance, acceleration [...]

Thanks, I get the meaning of using anchor in hdl_graph_slam. @koide3  By the way, I read the code of anchor in interactive_slam. I find that there is a little difference with hdl_graph_slam.  Could you explain that? For example, anchor node is set as first key frame's pose not the origin.

koide3, 2020-03-30:
In hdl_graph_slam, the anchor is completely fixed at the origin, and thus the first node will always be pulled to the origin and make some inconsistency. By moving the anchor to the current estimate of the first node, interactive_slam aims to solve this inconsistency. During optimization iterations, the anchor and the first node gradually move until they reach at the place where the first node would not have inconsistency with the other nodes.

koide3, 2020-03-30:
This video gives a good explanation. At every optimization, the anchor is moved to the current first node estimate so that it can settle to a good solution. 
https://drive.google.com/open?id=1a1Zi_3Lr6Wdj9nr54avyNASWWyRnogf4

The latest version of hdl_graph_slam also do this if "fix_first_node_adaptive" is enabled. 
https://github.com/koide3/hdl_graph_slam/blob/0331eec84007fc783aaaeb2fdab8d93cf7ef1a98/apps/hdl_graph_slam_nodelet.cpp#L581

wwtinwhu, 2020-03-30:
> This video gives a good explanation. At every optimization, the anchor is moved to the current first node estimate so that it can settle at a good solution.
> https://drive.google.com/open?id=1a1Zi_3Lr6Wdj9nr54avyNASWWyRnogf4
> 
> The latest version of hdl_graph_slam also do this if "fix_first_node_adaptive" is enabled.
> https://github.com/koide3/hdl_graph_slam/blob/0331eec84007fc783aaaeb2fdab8d93cf7ef1a98/apps/hdl_graph_slam_nodelet.cpp#L581

@koide3  Great explanation and implement for loose_fix! I think your code for this part helps me a lot! Thanks for your explanation and great job you have done!

closed 2020-03-30 by wwtinwhu