koide3/hdl_graph_slam #111: SLAM 3D using RoboSense Lidar

balletg, 2020-03-25:
Hi koide3,

Thanks for your open source project.
I am trying to use hdl to perform an hand held slam with the 16 nap rslidar.
I have modified the hdl_graph_slam.launch you have provided but each time I launch it I receive below error :

[FATAL] [1585131902.225136372]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `rslidar_nodelet_manager'
[prefiltering_nodelet-2] process has died [pid 9578, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet rslidar_nodelet_manager /rslidar_points:=/rslidar_points __name:=prefiltering_nodelet __log:=/home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/prefiltering_nodelet-2.log].
log file: /home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/prefiltering_nodelet-2*.log
[FATAL] [1585131902.496909072]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `rslidar_nodelet_manager'
[hdl_graph_slam_nodelet-4] process has died [pid 9585, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet rslidar_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/hdl_graph_slam_nodelet-4.log].
log file: /home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/hdl_graph_slam_nodelet-4*.log
[FATAL] [1585131902.753282107]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `rslidar_nodelet_manager'
[scan_matching_odometry_nodelet-3] process has died [pid 9579, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet rslidar_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/scan_matching_odometry_nodelet-3.log].
log file: /home/gabriel/.ros/log/196aa6d6-6e81-11ea-ab4f-40a3cc4085fd/scan_matching_odometry_nodelet-3*.log

Could you please help me understood why is it failing ?
I  am running it with ubuntu 18 and melodic and I have followed the instructions in the README to install hdl_graph_slam
Thanks

koide3, 2020-03-26:
Hi @balletg ,
Please make sure that you have installed ros-melodic-nodelet.

balletg, 2020-03-26:
Hi koide3,

balletg, 2020-03-26:
Hi koide3
Indeed it was not installed on one of my two computers (the one on which the lidar is connected to) but I am still having the same issue anyway