koide3/hdl_graph_slam #112: Outdoor SLAM

robot54, 2020-03-28:
Hi koide3,

First of all, let me thank you for your wonderful open source project!

I could apply your SLAM algorithm to a small indoor dataset captured by a Velodyne-HDL32E: I followed your recommandation and used an adapted version of "hdl_graph_slam_501.launch".
The results are impressive, so thank you again.

By now, I'm trying to apply your SLAM algorithm to an outdoor dataset captured by a Velodyne-HDL32E on top of a car (https://epan-utbm.github.io/utbm_robocar_dataset/). The total length of the set is some kilometers. Since the dataset is outdoor, I'm adapting  "hdl_graph_slam_400.launch". Unfortunately, I'm not able to get acceptable results, even after trying to modify most of the parameters. The odometry estimation indeed introduces a large drit, causing the algorithm to miss the loop closures. I think I could increase the "distance_tresh" parameter, but mainly for computational reasons, I would prefer to improve the scan-matching odometry.

Have you succeeded in applying your algorithm to large outdoor datasets?
And, if you have, have you some advices?

Have a nice weekend

koide3, 2020-03-30:
Hi @robot54 ,

I think you can give GICP a try. I found that NDT shows a large drift in a large environment while GICP is much accurate in many environments (though it is more time consuming).

If you are on ROS melodic, you can just change "registration_method" in your launch file to "GICP" (GICP in kinetic has a bug). You can also try our fast GICP implementations by adding them to "registrations.cpp".

https://github.com/SMRT-AIST/fast_gicp
https://github.com/koide3/hdl_graph_slam/blob/master/src/hdl_graph_slam/registrations.cpp

robot54, 2020-03-31:
Hi @koide3 ,

Thank you very much for your answer.
I forgot to mention that I'm using ROS kinetic. Hence, as you said, I can't use GICP (or your new fast implementations). I tried with "GICP_omp". Due to the higher computational load, I had to reduce the publish rate of the rosbag file, but the results are indeed better (even though they're not satisfying yet).

Are you aware of any efficient registration methods that could further reduce the drift in large environments and that work in ROS kinetic?

Have a nice day.