koide3/hdl_graph_slam #113: map frame is transformed against odom instead of the opposite

ricber, 2020-04-04:
Hi @koide3 ,
I have noticed that the loop closure corrections shift the map frame against the odom frame? Is this by design? Usually, the map frame is static and the correction from loop closure is applied shifting the odom frame. Shifting the map frame cause the final map to have a weird translation and rotation relative to the map frame.

koide3, 2020-04-06:
Hi @ricber ,
What I implemented is something similar to the one in amcl, in which the ```map-odom``` transformation compensates for odometry errors and keeps ```base_link``` in the correct position on the map (see 3.1.6 at http://wiki.ros.org/amcl). So, shifting ```map``` against ```odom``` itself is by design. Everytime a new constraint is added, the ```map``` frame can be shifted against the ```odom``` frame. 

But, the ```map``` frame should not be shifted against the ```odom``` frame when no constraints are made. If you see such a shifting phenomenon, there may be a problem in the ```map-odom``` transformation handling. I try to fix this problem.

ricber, 2020-04-06:
When you say that no constraints are made... isn't the loop closure a constraint added to the pose graph? Thus, it is normal to see a shift in the transformation between map and odom. 
Anyway, why can't you just sve the map relative to odom frame? This will also ease the task of giving the initial position of the robot to the localization lagorithm. If the odom frame stays fixed in the initial position, thus, the initial positon for the UKF will always be the origin.