koide3/hdl_graph_slam #119: TF error: [Lookup would require extrapolation into the future. Requested time 1509348922.367725000 but the latest data is at time 1509348882.233323000, when looking up transform from frame [velodyne] to frame [map]]

Tengyun-Mo, 2020-04-16:
The map of point cloud can't display well when I use rviz.And I get error:
`TF error: [Lookup would require extrapolation into the future. Requested time 1509348922.367725000 but the latest data is at time 1509348882.233323000, when looking up transform from frame [velodyne] to frame [map]]`
How to resolve this issue?Who can help me?I'm very appreciate it .

ASE-HHN, 2020-04-17:
I guess that's an error when you don't execute "set use_sim_time true" before starting the launch file.

Tengyun-Mo, 2020-04-19:
> I guess that's an error when you don't execute "set use_sim_time true" before starting the launch file.

Thanks for your response,I have executed "set use_sim_time true" before start the launch file.And this error never changed.

ASE-HHN, 2020-04-19:
Have you also started the bagfile with the clock setting, like "rosbag play --clock /home/your/path/*.bag"?

Tengyun-Mo, 2020-04-19:
> Have you also started the bagfile with the clock setting, like "rosbag play --clock /home/your/path/*.bag"?
Yes,I follow every step of reference

Tengyun-Mo, 2020-04-19:
[image]
This is the map of cloud point after played the bag file.It's just a few laser scan have been used.Everything is OK after played bag file,but when I playing bag file ,the error occurred.

mhaboali, 2020-05-15:
I have the same problem, any help with it?

goat-spike, 2023-02-07:
I solved this problem. I ran this package on jetson agx xavier and met the same problem. I think it is because the play speed of the bag is too fast. so when i use 

> rosbag play --clock hdl_400.bag -r 0.5
it is working。Maybe the jetson agx xavier use this package there has the hardware limitation